nordicsemi/Android-nRF-Mesh-Library #239: hello，Messages always accept errors

xiaofengming, 2019-08-27:
DefaultNoOperationMessageState: Message reassembly may not be completed yet!

roshanrajaratnam, 2019-08-27:
Hi sorry about this, could you elaborate the issue please?

xiaofengming, 2019-08-27:
not solved ！ The message can be done, but I think the return data is too slow，
I want to accept the message quickly.

WangManAndroid, 2019-09-03:
我的用的就很正常啊  你是哪家公司？

WangManAndroid, 2019-09-03:
你看看是否是你们硬件协议栈问题    我用的就很正常

xiaofengming, 2019-09-04:
深圳一家公司，循环设置ConfigModel 类的消息（比如BoundAppKey ，PublicationSet），你有收不到的情况吗？

WangManAndroid, 2019-09-05:
> 我的用的就很正常啊 你是哪家公司？
正常的ConfigMessage  GenericMessage 都会在150+（50*TTL）ms  收到，如果经常收不到你的抓包看看了，很有可能是你们硬件协议栈问题。我这边也会有时收不到相应的Status，但是很少很少。

WangManAndroid, 2019-09-05:
case MeshManagerApi.PDU_TYPE_NETWORK:

                    //This is not possible however let's return null
                    if (provisionerAddress == null) {
                        return null;
                    }

                    //Check if the message is directed to us, if its not ignore the message
                    if (provisionerAddress != dst) {
                        Log.v(TAG, "Received a control message that was not directed to us, so we drop it");
                        return null;
                    }

                    if (isSegmentedMessage(decryptedNetworkPayload[2])) {
                        return parseSegmentedControlMessage(data, decryptedProxyPdu, ttl, src, dst);
                    } else {
                        return parseUnsegmentedControlMessage(data, decryptedProxyPdu, ttl, src, dst, sequenceNumber);
                    }

这是最新的代码  你看看跟你的是否不同  NetWorkLayer

xiaofengming, 2019-09-10:
Gatt链接的设备木有回不来情况，周围mesh设备config类型消息很大概率回不来

xiaofengming, 2019-09-11:
@roshanrajaratnam hello，The mesh device is not a gatt device, and there is always a situation where ConfigMessage cannot receive it.

roshanrajaratnam, 2019-09-11:
Hi @xiaofengming what do you mean by the mesh device is not a gatt device? If the device supports the Mesh Provisioning Service and Proxy Service it will can communicate over the GATT bearer. Currently our sample application does not have the Advertising bearer implemented. The users of this library are welcome to implement their own advertising bearer and use the library as it is. Hope this answers your question :)

xiaofengming, 2019-09-11:
@roshanrajaratnam The proxy device will not accept the ConfigMessage failure. The device forwarded by the proxy will accept the ConfigMessage failure, not 100%, but it is very frequent.

roshanrajaratnam, 2019-09-11:
@xiaofengming have you checked the TTL to make sure if the message can reach the end device via the proxy?

xiaofengming, 2019-09-12:
globalTtl = 5 ，node  publishTtl =16；

WangManAndroid, 2019-09-24:
@xiaofengming   这个你得debug,然后发现问题，定位问题原因，我怀疑你们硬件mesh 协议栈问题，我现在开发就用的这个nordic 的demo，有小bug ，有些硬件协议栈实现问题，你得仔细debug,我也是一步一步摸索出来的，最总都能定位是什么问题，都能解决。

closed 2019-10-11 by xiaofengming