nordicsemi/Android-nRF-Mesh-Library #243: BluetoothGatt: Unhandled exception in callback

WangManAndroid, 2019-09-05:
9-03 20:43:35.880 23061-23120/no.nordicsemi.android.nrfmeluetoothGatt: onCharacteristicWrite() - Device=11:11:11:1000-1000-8000-00805f9b34fb Status=09-03 20:43:35.880 23061-23120/no.nordicsemi.android.nrfmeleMeshManager: Data written: 0xC05EC309FE8ADA509-03 20:43:35.880 23061-23120/no.nordicsemi.android.nrfmeluetoothGatt: Unhandled exception in callback java.util.NoSuchElementException at java.util.LinkedList.removeFirstImpl(LinkedList at java.util.LinkedList.removeFirst(LinkedList.jav at java.util.LinkedList.poll(LinkedList.java:895) at no.nordicsemi.android.ble.BleManager$BleManagerleManager.java:1518) at no.nordicsemi.android.ble.BleManager$BleManagernCharacteristicWrite(BleManager.java:1349) at android.bluetooth.BluetoothGatt$1.onCharacterisava:388) at android.bluetooth.IBluetoothGattCallback$Stub.oBluetoothGattCallback.java:279) at android.os.Binder.execTransact(Binder.java:404) at dalvik.system.NativeStart.run(Native Method)

roshanrajaratnam, 2019-09-05:
@WangManAndroid This looks liek an issue in the Bluetooth Library. Have you changed anything here?

WangManAndroid, 2019-09-06:
`/*
 * Copyright (c) 2015, Nordic Semiconductor
 * All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without modification, are permitted provided that the following conditions are met:
 *
 * 1. Redistributions of source code must retain the above copyright notice, this list of conditions and the following disclaimer.
 *
 * 2. Redistributions in binary form must reproduce the above copyright notice, this list of conditions and the following disclaimer in the
 * documentation and/or other materials provided with the distribution.
 *
 * 3. Neither the name of the copyright holder nor the names of its contributors may be used to endorse or promote products derived from this
 * software without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
 * LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT
 * HOLDER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT
 * LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON
 * ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE
 * USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 */
package no.nordicsemi.android.ble;

import android.annotation.TargetApi;
import android.bluetooth.BluetoothAdapter;
import android.bluetooth.BluetoothDevice;
import android.bluetooth.BluetoothGatt;
import android.bluetooth.BluetoothGattCallback;
import android.bluetooth.BluetoothGattCharacteristic;
import android.bluetooth.BluetoothGattDescriptor;
import android.bluetooth.BluetoothGattService;
import android.bluetooth.BluetoothProfile;
import android.content.BroadcastReceiver;
import android.content.Context;
import android.content.Intent;
import android.content.IntentFilter;
import android.os.Build;
import android.os.Handler;
import android.os.Looper;
import android.support.annotation.NonNull;
import android.support.annotation.Nullable;
import android.support.annotation.RequiresApi;
import android.support.annotation.StringRes;
import android.util.Log;

import java.lang.reflect.Method;
import java.util.Deque;
import java.util.LinkedList;
import java.util.Queue;
import java.util.UUID;
import java.util.concurrent.ConcurrentLinkedQueue;

import no.nordicsemi.android.ble.error.GattError;
import no.nordicsemi.android.ble.utils.BleConnectionCompat;
import no.nordicsemi.android.ble.utils.ILogger;
import no.nordicsemi.android.ble.utils.ParserUtils;
import no.nordicsemi.android.log.ILogSession;
import no.nordicsemi.android.log.Logger;

// DO NOT EDIT THIS FILE UNLESS NECESSARY!
/**
 * <p>
 * The BleManager is responsible for managing the low level communication with a Bluetooth LE device.
 * Please see profiles implementation in Android nRF Blinky or Android nRF Toolbox app for an
 * example of use.
 * <p>
 * This base manager has been tested against number of devices and samples from Nordic SDK.
 * <p>
 * The manager handles connection events and initializes the device after establishing the connection.
 * <ol>
[... 1641 lines cut ...]
			default:
				return "UNKNOWN: " + type;
		}
	}

	/**
	 * Converts the connection state to String value
	 *
	 * @param state the connection state
	 * @return state as String
	 */
	protected String stateToString(final int state) {
		switch (state) {
			case BluetoothProfile.STATE_CONNECTED:
				return "CONNECTED";
			case BluetoothProfile.STATE_CONNECTING:
				return "CONNECTING";
			case BluetoothProfile.STATE_DISCONNECTING:
				return "DISCONNECTING";
			default:
				return "DISCONNECTED";
		}
	}
}
`

WangManAndroid, 2019-09-06:
Is there concurrency in this library？

closed 2019-10-30 by roshanrajaratnam