nordicsemi/Android-nRF-Mesh-Library #246: BLE connections often offline

zhangping001, 2019-09-10:
**After each successful connection, about a minute to disconnect! The following is the log:**

09-10 08:49:43.477 13955-13955/no.nordicsemi.android.nrfmeshprovisioner V/BleManager: Connecting...
09-10 08:49:43.477 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BleManager: gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE)
09-10 08:49:43.478 13955-13955/no.nordicsemi.android.nrfmeshprovisioner I/BluetoothDevice: connectGatt: 2
09-10 08:49:43.479 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: connect() - device: 1B:D7:BC:07:DA:78, auto: false
09-10 08:49:43.479 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: registerApp()
09-10 08:49:43.481 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: registerApp() - UUID=341ea224-db52-4578-b5d9-0cfbbd685c56
09-10 08:49:43.495 13955-14168/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
09-10 08:49:46.427 13955-13967/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=1B:D7:BC:07:DA:78
09-10 08:49:46.438 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BleManager: [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
09-10 08:49:46.438 13955-13955/no.nordicsemi.android.nrfmeshprovisioner I/BleManager: Connected to 1B:D7:BC:07:DA:78
09-10 08:49:46.440 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BleManager: wait(300)
09-10 08:49:46.757 13955-13955/no.nordicsemi.android.nrfmeshprovisioner V/BleManager: Discovering services...
09-10 08:49:46.757 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BleManager: gatt.discoverServices()
09-10 08:49:46.757 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: discoverServices() - device: 1B:D7:BC:07:DA:78
09-10 08:49:46.771 13955-13966/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onSearchComplete() = Device=1B:D7:BC:07:DA:78 Status=0
09-10 08:49:46.784 13955-13955/no.nordicsemi.android.nrfmeshprovisioner I/BleManager: Services discovered
09-10 08:49:46.784 13955-13955/no.nordicsemi.android.nrfmeshprovisioner V/BleManager: Primary service found
09-10 08:49:46.784 13955-13955/no.nordicsemi.android.nrfmeshprovisioner V/BleManager: Requesting new MTU...
09-10 08:49:46.784 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BleManager: gatt.requestMtu(517)
09-10 08:49:46.784 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: configureMTU() - device: 1B:D7:BC:07:DA:78 mtu: 517
09-10 08:49:46.851 13955-13967/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onConfigureMTU() - Device=1B:D7:BC:07:DA:78 mtu=131 status=0
09-10 08:49:46.868 13955-13955/no.nordicsemi.android.nrfmeshprovisioner I/BleManager: MTU changed to: 131
09-10 08:49:46.868 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BleManager: gatt.setCharacteristicNotification(00002adc-0000-1000-8000-00805f9b34fb, true)
09-10 08:49:46.868 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002adc-0000-1000-8000-00805f9b34fb enable: true
09-10 08:49:46.871 13955-13955/no.nordicsemi.android.nrfmeshprovisioner V/BleManager: Enabling notifications for 00002adc-0000-1000-8000-00805f9b34fb
09-10 08:49:46.871 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BleManager: gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
09-10 08:49:46.951 13955-13955/no.nordicsemi.android.nrfmeshprovisioner I/BleManager: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
09-10 08:49:46.951 13955-13955/no.nordicsemi.android.nrfmeshprovisioner I/BleManager: Notifications enabled
09-10 08:49:46.951 13955-13955/no.nordicsemi.android.nrfmeshprovisioner V/BleManager: Reading characteristic 00002a19-0000-1000-8000-00805f9b34fb
09-10 08:49:46.951 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BleManager: gatt.readCharacteristic(00002a19-0000-1000-8000-00805f9b34fb)
09-10 08:49:47.101 13955-13955/no.nordicsemi.android.nrfmeshprovisioner I/BleManager: Read Response received from 00002a19-0000-1000-8000-00805f9b34fb, value: (0x) 00
09-10 08:49:47.102 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BleMeshManager: onCharacteristicRead...
09-10 08:49:47.102 13955-13955/no.nordicsemi.android.nrfmeshprovisioner I/BleManager: Battery Level received: 0%
09-10 08:50:46.234 13955-14169/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: onClientConnectionState() - status=19 clientIf=6 device=1B:D7:BC:07:DA:78
09-10 08:50:46.235 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BleManager: [Callback] Connection state changed with status: 19 and new state: 0 (DISCONNECTED)
09-10 08:50:46.236 13955-13955/no.nordicsemi.android.nrfmeshprovisioner W/BleManager: Error: (0x13): GATT CONN TERMINATE PEER USER
09-10 08:50:46.236 13955-13955/no.nordicsemi.android.nrfmeshprovisioner I/BleManager: Disconnected
09-10 08:50:46.238 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BleManager: gatt.close()
09-10 08:50:46.238 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: close()
09-10 08:50:46.238 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothGatt: unregisterApp() - mClientIf=6
09-10 08:50:46.244 13955-13955/no.nordicsemi.android.nrfmeshprovisioner V/NrfMeshRepository: Disconnected
09-10 08:50:46.305 13955-13955/no.nordicsemi.android.nrfmeshprovisioner I/BluetoothAdapter: getBluetoothLeScanner
09-10 08:50:46.310 13955-13955/no.nordicsemi.android.nrfmeshprovisioner I/BluetoothLeScanner: startScan 3 is called
09-10 08:50:46.311 13955-13955/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothAdapter: STATE_ON
09-10 08:50:46.321 13955-14168/no.nordicsemi.android.nrfmeshprovisioner D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6

roshanrajaratnam, 2019-09-10:
Hi @zhangping001 if you look at the logs the error is 0x13, which means the Gatt connection was terminated by the peripheral.

roshanrajaratnam, 2019-09-25:
Closing this due to inactivity, but please feel free to reopen again.

closed 2019-09-25 by roshanrajaratnam