nordicsemi/Android-nRF-Mesh-Library #247: MeshMessageState: Incomplete timer has expired, all segments were not received!

xiaofengming, 2019-09-20:
hello ,@roshanrajaratnam 
2019-09-20 17:38:45.498 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: Received a segmented access message from: 0002
2019-09-20 17:38:45.498 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: SEG O: 0
2019-09-20 17:38:45.498 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: SEG N: 5
2019-09-20 17:38:45.498 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: Received segment message count: 5
2019-09-20 17:38:45.993 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: Received a segmented access message from: 0002
2019-09-20 17:38:45.993 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: SEG O: 1
2019-09-20 17:38:45.993 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: SEG N: 5
2019-09-20 17:38:45.994 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: Received segment message count: 5
2019-09-20 17:38:46.736 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: Received a segmented access message from: 0002
2019-09-20 17:38:46.736 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: SEG O: 2
2019-09-20 17:38:46.736 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: SEG N: 5
2019-09-20 17:38:46.736 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: Received segment message count: 5
2019-09-20 17:38:47.232 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: Received a segmented access message from: 0002
2019-09-20 17:38:47.232 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: SEG O: 0
2019-09-20 17:38:47.232 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: SEG N: 5
2019-09-20 17:38:47.232 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: Received segment message count: 5
2019-09-20 17:38:47.973 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: Received a segmented access message from: 0002
2019-09-20 17:38:47.973 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: SEG O: 1
2019-09-20 17:38:47.973 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: SEG N: 5
2019-09-20 17:38:47.973 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: Received segment message count: 5
2019-09-20 17:38:48.468 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/NetworkLayer: Received a segmented access message from: 0002
2019-09-20 17:38:48.468 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: SEG O: 2
2019-09-20 17:38:48.468 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: SEG N: 5
2019-09-20 17:38:48.469 19450-19462/no.nordicsemi.android.nrfmeshprovisioner V/LowerTransportLayer: Received segment message count: 5
2019-09-20 17:38:58.479 19450-19450/no.nordicsemi.android.nrfmeshprovisioner V/MeshMessageState: Incomplete timer has expired, all segments were not received!

WangManAndroid, 2019-09-23:
@xiaofengming   你们的mesh 协议栈是自己开发的么？我使用就很正常，像这种问题你们应该跟你们协议栈开发人员 沟通。

xiaofengming, 2019-09-23:
嗯嗯，谢谢！

closed 2019-10-30 by xiaofengming