nordicsemi/Android-nRF-Mesh-Library #248: Failed to send message to node after importing network

NIUNIUN, 2019-09-23:
---

**Describe the bug**
With the export/import network feature, a message cannot be sent to a node after the network is imported.

**Platform details:**
 - Device: Replay: src 0x0001 dst 0xc000 seq 0x00009c.

**Logs / Screenshots**
[image]

WangManAndroid, 2019-09-23:
看这个消息应该是分组消息，你看看蓝牙硬件打印,调试一下

WangManAndroid, 2019-09-23:
@NIUNIUN   你试试看改一下src地址，有可能是重放保护了

roshanrajaratnam, 2019-09-23:
@NIUNIUN After exporting make sure you change the provisiner address as the sequence numbers are not shared the messages sent from the same address on a different phone will fail.

NIUNIUN, 2019-09-24:
@roshanrajaratnam thank you!

closed 2019-09-24 by NIUNIUN

roshanrajaratnam, 2019-09-24:
@NIUNIUN You're welcome!