nordicsemi/Android-nRF-Mesh-Library #249: NullPointerException when getting UUID

hariharasudhanPerumalSamy, 2019-09-23:
---

**Describe the bug**
we try to provision the nrf52840-DK module, with android nordic library version 2.0.5.
sometimes we are getting NullPointer Exception in the middle of the provisioning from ProvisionedMeshNode.java where it tries to get the UUID which gives null.

**Logs / Screenshots**
D/STEPS : ( 8571): PROVISIONING_DATA_SENT
D/OzeoMeshManagerCallbacks( 8571): getMtu
D/OzeoMeshManagerCallbacks( 8571): sendProvisioningPdu
V/BleMeshManager( 8571): Sending data : 0x0307D77EEEA7A20A699E9F8B461893E257363AE45A2F6C092DADA250359288A4A9A73F
V/BleMeshManager( 8571): Data written: 0x0307D77EEEA7A20A699E9F8B461893E257363AE45A2F6C092DADA250359288A4A9A73F
D/OzeoUnprovisionedGattCallbacks( 8571): onDataSent
V/MeshManagerApi( 8571): Provisioning pdu sent: 0x0307D77EEEA7A20A699E9F8B461893E257363AE45A2F6C092DADA250359288A4A9A73F
I/flutter ( 8571): event received : {data: unknowState, type: provisioningStateChanged}
I/flutter ( 8571): [NrfMeshService] _onEvent not handled (unknown type: provisioningStateChanged)
V/BleMeshManager( 8571): Characteristic notified: 0x0308
D/OzeoUnprovisionedGattCallbacks( 8571): onDataReceived
V/MeshManagerApi( 8571): Received provisioning message: 0x0308
W/BluetoothGatt( 8571): Unhandled exception in callback
W/BluetoothGatt( 8571): java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String java.util.UUID.toString()' on a null object reference
W/BluetoothGatt( 8571): 	at no.nordicsemi.android.meshprovisioner.transport.ProvisionedMeshNode.<init>(ProvisionedMeshNode.java:87)
W/BluetoothGatt( 8571): 	at no.nordicsemi.android.meshprovisioner.MeshProvisioningHandler.parseProvisioningState(MeshProvisioningHandler.java:189)
W/BluetoothGatt( 8571): 	at no.nordicsemi.android.meshprovisioner.MeshProvisioningHandler.parseProvisioningNotifications(MeshProvisioningHandler.java:153)
W/BluetoothGatt( 8571): 	at no.nordicsemi.android.meshprovisioner.MeshManagerApi.parseNotifications(MeshManagerApi.java:376)
W/BluetoothGatt( 8571): 	at no.nordicsemi.android.meshprovisioner.MeshManagerApi.handleNotifications(MeshManagerApi.java:329)
W/BluetoothGatt( 8571): 	at fr.nocoffee.ozeoFlutter.OzeoUnprovisionedGattCallbacks.onDataReceived(OzeoUnprovisionedGattCallbacks.java:29)
W/BluetoothGatt( 8571): 	at fr.nocoffee.ozeoFlutter.BleMeshManager$1.onCharacteristicNotified(BleMeshManager.java:172)
W/BluetoothGatt( 8571): 	at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.onCharacteristicChanged(BleManager.java:1440)
W/BluetoothGatt( 8571): 	at android.bluetooth.BluetoothGatt$1$8.run(BluetoothGatt.java:468)
W/BluetoothGatt( 8571): 	at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:770)
W/BluetoothGatt( 8571): 	at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:39)
W/BluetoothGatt( 8571): 	at android.bluetooth.BluetoothGatt$1.onNotify(BluetoothGatt.java:462)
W/BluetoothGatt( 8571): 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:185)
W/BluetoothGatt( 8571): 	at android.os.Binder.execTransact(Binder.java:752)
Application finished.

roshanrajaratnam, 2019-09-24:
@hariharasudhanPerumalSamy If you are using the library, please make sure you refresh the device gatt cache as Android uses the cache to avoid service discovery therefore the device will not find the Proxy service after the provisioning is complete. However this might be different from Android version 8 onwards if I am not wrong. In our sample app we do a refresh the gatt cache to avoid this.

closed 2019-09-24 by roshanrajaratnam

hariharasudhanPerumalSamy, 2019-09-30:
@roshanrajaratnam i referred the sample app, as you mentioned above, you are refreshing the gatt cache on provisioning complete and so we are. but this null exception is happening during provisioning.