nordicsemi/Android-nRF-Mesh-Library #250: Public key response mismatch when switching no.nordicsemi.android:ble dependency version

WuLFGit, 2019-09-24:
---

I tried to provision a device multiple times failing using no.nordicsemi.android:ble:2.1.1 version. Switching back to no.nordicsemi.android:ble:1.2.0 makes provisioning work back again. What i noticed is a difference between response byte array i receive when sending public key to my device. 

May i ask you why this behaviour happens? It sounds weird to me that switching ble library version result in different byte array response received from the device.

**Logs / Screenshots**
Here some screenshot about responses i receive when public key is sent:

Public key sent response on no.nordicsemi.android:ble:2.1.1. As you can notice, on position 1 i expect value 3 but i receive -99:
[image]

Public key sent response on no.nordicsemi.android:ble:1.2.0. As you can notice, on position 1 there is the expected value 3:
[image]

Responses length is different also.

Thank you in advance for you support.

roshanrajaratnam, 2019-09-25:
It indeed looks different and it should not do that., Are you looking at the same responses?

WuLFGit, 2019-09-25:
Yes, i double checked and the flow i follow is the exact same thing when switching from old ble version to new one.
Provisioning states that are triggered in parseProvisioningNotifications are in order:

- PROVISIONING_CAPABILITIES
- PROVISIONING_START
- PROVISIONING_PUBLIC_KEY

inside PROVISIONING_PUBLIC_KEY, validateMessage is successful with the old version, but unsuccesful with the new one, because byte arrays are different. Sounds weird to me.

roshanrajaratnam, 2019-10-03:
@WuLFGit I'm currently busy with some other stuff at the moment. So you might want to continue your development with Ble Library 1.2. This is the same reason why I have mentioned the recomeded version is 1.2 as we have not tested the mesh library with 2.1. However it should ideally work out of the box.

philips77, 2019-10-22:
Hi @WuLFGit,
I'm the author of Android BLE Libraries (both 1.2 and 2.x). Could you enable logging in the BLE library (override the `log` method in the `BleManager`) 2.0 and send us the logs? Version 1.2 logs data in nRF Logger when you set the LogSession, this could also be helpful. Thank you.

philips77, 2019-11-08:
Hi, could you have a look at #269? We migrated this project to BLE Library 2.1.1 and could not replicate your issue.

lenix1988, 2019-12-25:
@philips77 
I found a bug that if proxy segment PDU is more than two then app will stuck in provision because app parse proxy segment PDU failed.
[image]

so we should update provision proxy segment PDU process function
Reference Code(I am not good at Java programming ):
```java
    private byte[] appendPdu(final int mtuSize, final byte[] pdu) {
        if (mIncomingBuffer == null) {
            final int length = Math.min(pdu.length, mtuSize);
            mIncomingBufferOffset = 0;
            mIncomingBufferOffset += length;
            byte buf =  (byte)(MeshParserUtils.unsignedByteToInt(pdu[0])& 0x3f);
            //Make sure  Provision PDU  header to be unsegment PDU header for following function
            pdu[0] = buf;
            mIncomingBuffer = pdu;
            Log.v(TAG, " PDU head type： "+  pdu[0]);

        } else {
            //length remove proxy header
            final int length = (Math.min(pdu.length, mtuSize) - 1);
            final byte[] buffer = new byte[mIncomingBuffer.length + length];
            System.arraycopy(mIncomingBuffer, 0, buffer, 0, mIncomingBufferOffset);
            //copy after proxy header
            System.arraycopy(pdu, 1, buffer, mIncomingBufferOffset, length);
            mIncomingBufferOffset += length;
            mIncomingBuffer = buffer;

            final int pduType = (MeshParserUtils.unsignedByteToInt(pdu[0]) >> SAR_BIT_OFFSET);
            if ( pduType == GATT_SAR_END )
            {
//                if (mIncomingBufferOffset < mtuSize) {
                Log.v(TAG, " mIncomingBuffer Size： "+ mIncomingBuffer.length);
                final byte[] packet = mIncomingBuffer;
                mIncomingBuffer = null;
                return packet;
//                }
            }
        }
        return null;
    }
```
[image]

GeekKernel, 2020-03-05:
> @philips77
> I found a bug that if number of proxy segment PDU for one proxy PDU is more than two then app will stuck in Mesh Device sended public key because app parse proxy segment PDU failed.
> [image]
> 
> so we should update provision proxy segment PDU process function
> Reference Code(I am not good at Java programming ):
> 
> ```java
>     private byte[] appendPdu(final int mtuSize, final byte[] pdu) {
>         if (mIncomingBuffer == null) {
>             final int length = Math.min(pdu.length, mtuSize);
>             mIncomingBufferOffset = 0;
>             mIncomingBufferOffset += length;
>             byte buf =  (byte)(MeshParserUtils.unsignedByteToInt(pdu[0])& 0x3f);
>             //Make sure  Provision PDU  header to be unsegment PDU header for following function
>             pdu[0] = buf;
>             mIncomingBuffer = pdu;
>             Log.v(TAG, " PDU head type： "+  pdu[0]);
> 
>         } else {
>             //length remove proxy header
>             final int length = (Math.min(pdu.length, mtuSize) - 1);
>             final byte[] buffer = new byte[mIncomingBuffer.length + length];
>             System.arraycopy(mIncomingBuffer, 0, buffer, 0, mIncomingBufferOffset);
>             //copy after proxy header
>             System.arraycopy(pdu, 1, buffer, mIncomingBufferOffset, length);
>             mIncomingBufferOffset += length;
>             mIncomingBuffer = buffer;
> 
>             final int pduType = (MeshParserUtils.unsignedByteToInt(pdu[0]) >> SAR_BIT_OFFSET);
>             if ( pduType == GATT_SAR_END )
>             {
> //                if (mIncomingBufferOffset < mtuSize) {
>                 Log.v(TAG, " mIncomingBuffer Size： "+ mIncomingBuffer.length);
>                 final byte[] packet = mIncomingBuffer;
>                 mIncomingBuffer = null;
>                 return packet;
> //                }
>             }
>         }
>         return null;
>     }
> ```
> 
> [image]

This is my problem! Android mesh library cannot process more than 2 proxy segment PDUs,but if ATT MTU is 69  octets ,it's enough to store Public Key data。The foremost is ATT MTU exchange did'nt work correctly. I'm sorry,please close my comment.

roshanrajaratnam, 2020-03-05:
Hi Thanks for reporting this issue, will look in to this.

closed 2020-06-08 by roshanrajaratnam