nordicsemi/Android-nRF-Mesh-Library #251: Sub-mesh networks

WuLFGit, 2019-09-24:
**Is your feature request related to a problem? Please describe.**
In my app i've two requirements:

- to have multiple mesh networks on the same device
- to backup and restore mesh Room database from cloud

**Describe the solution you'd like**
For the first point, is that already possible? If yes, what's the best practice on how to do it with Android-nRF-Mesh-Library structure?

For the second point, i would like to save what's needed to restore a whole mesh network (or multiple ones) from one device to another without doing the whole provisioning process etc. Is that possible?

Thank you!

roshanrajaratnam, 2019-09-25:
Hi @WuLFGit, the room databse is designed to handle multiple mesh networks but I have not tested this and there is no api implemented in the mesh library to achieve this as of now. This is something that is already in the backlog so we will add this sometime down the line.

Regarding the second point the importing/exporting a mesh network should allow you to restore the network on different devices(multiple ones would require exporting one by one). This way only one device can do the provisioning and the rest can just impor the network.

WuLFGit, 2019-09-25:
Ok that makes sense.
The only thing that comes to my mind is: What if i have to add a new node on the new device that may be on a different OS? As i read in iOS related post, when we import mesh network on a new device we should create a new Provisioner. Is this really necessary?

Also, what's really needed to save to make sure i can re-create a Provisioner on the new device?

roshanrajaratnam, 2019-09-25:
Well this is a requirement by mesh specification and is not related to an OS. Each message sent by a node has a uniqure sequence number that increments which is bound to the address of that node. So a new node means a new address which also means a neq sequence number. There are two approaches here, either you can have  a new provisioner or the second approach would be, change the address of that privisioner after importing a mesh network. I would go with the first approach to avoid confusion.

WuLFGit, 2019-09-25:
Sure. So we can create a new Provisioner to keep provisioning nodes on the same network. I guess i can create the new provisioner with same AppKey and NetKey i used before, right?

WuLFGit, 2019-09-25:
Ok, so i guess that a phone device which provisions other nodes is represented as a Provisioner in the mesh network. So to make sure the whole mesh network works on both devices the cleaner approach would be what you suggested. My next question then is: can i create a new provisioner with same AppKeys and NetKeys?

Btw, thank you for you support and quick responses!

roshanrajaratnam, 2019-09-25:
Yes you can do this, because a mesh network can have multiple provisioners. Creating a provisioner will always give access to all the keys available in that network

closed 2020-04-16 by roshanrajaratnam