nordicsemi/Android-nRF-Mesh-Library #252: Quick start / Tutorial

isactornberg, 2019-09-24:
Hey!

I am currently trying to port this as well as the iOS-nRF-Mesh-Library to react-native but I am having a rough time picking out the "important/main" parts from the sample and library.

In the readme it's mentioned "_To start using the library in your own project take a look at the following snippet_." but it doesn't seem to contain any information on how to get started with scanning, provisioning and sending/receiving data. Implementing the interfaces for the callbacks is a good start but finding what root classes I should use to make basic calls isn't seems unnecessarily hard. Would love some for pointers for what classes should be used to create a nice bridge to the react layer!

Also there seem to be a lot of logic in the sample class but is any of this necessary to get started with some basic and naive communication or should I just go with the meshprovider and then create the necessary logic on the react layer?

Happy for any help I can get :smiley:

roshanrajaratnam, 2019-09-25:
Hi @isactornberg, if I were you I would not pick adn choose classes to port it. Instead I would try to implement a wrapper based on the callbacks and the api provided by the mesh library. If not you will end up implementing the whole stack in react.

isactornberg, 2019-09-25:
Hi @roshanrajaratnam, thanks for your reply.

That sounds like the proper way to go but I feel like the MeshManagerApi with the callback listeners set still lacks some functionality, but I might probably be wrong here.

E.g if I wish to start scanning for unprovisioned nodes and then provision one of them how would I use the API in this case? I can see there are different types of provisioning functions but before using one of these functions it requires me to find an unprovisioned node. Maybe this done automatically in the loadNetwork() and I haven't digged through everything enough yet. 

But basically the MeshManagerApi with all the callbacks should be enough to use for the wrapper?

roshanrajaratnam, 2019-09-25:
Hi, so the mesh library is just transport independent. It does not have any scanning or bluetooth connectivity. This is something we have left out from the mesh library allowing users to decide to use a preferred way of scanning and connecting. The example app uses our Scanner-Compat library as well as our BLE library v1.2.0.

philips77, 2019-09-25:
Or BLE Library 2.x.

isactornberg, 2019-09-25:
Ah okey then I follow. :+1: 

Then I guess you will do all the scanning/writing/reading using the ble library of your choice. How do you interface this with the meshprovisioner? Do you pass data from and to the ble library to be able to read/write information or is there a more straight forward way?

This might be outside this scope and I can probably dig through the example app to see how this works but will appreciate all the help I can get to get started. :smile:

roshanrajaratnam, 2019-09-25:
Just like you mentioned, its a pretty simple approach we pass data from the ble library to the mesh library. Basically everytime a notificaiton is received this is sent to mesh library. Hope this helps you getting started ;)

closed 2019-10-07 by roshanrajaratnam

bdent, 2020-02-10:
Hi @isactornberg,

How far did you manage to get with this?

I'm looking to do the same.

Do you have an email I can reach out to you at?

Thanks.

isactornberg, 2020-02-12:
Hey @bdent!

Put this sorta on hold first but now I am working on a flutter integration that I have sorta gotten working with limited feature set, you can reach out to me at isac.tornberg@gmail.com. :)