nordicsemi/Android-nRF-Mesh-Library #257: the isValidUnassignedAddress() method is incorrect

xiaofengming, 2019-10-15:
@roshanrajaratnam 
/**
     * Validates an unassigned address
     *
     * @param address 16-bit address
     * @return true if the address is a valid unassigned address or false otherwise
     */
    public static boolean isValidUnassignedAddress(@NonNull final byte[] address) {
        if (isAddressInRange(address)) {
            return false;
        }
        return isValidUnassignedAddress(MeshParserUtils.unsignedBytesToInt(address[0], address[1]));
    }

--------------------Should be changed

/**
     * Validates an unassigned address
     *
     * @param address 16-bit address
     * @return true if the address is a valid unassigned address or false otherwise
     */
    public static boolean isValidUnassignedAddress(@NonNull final byte[] address) {
        if (!isAddressInRange(address)) {
            return false;
        }
        return isValidUnassignedAddress(MeshParserUtils.unsignedBytesToInt(address[0], address[1]));
    }

roshanrajaratnam, 2019-10-18:
Nice catch, I'll be out of work for the next 2 weeks and will fix when I am back. Also free to contribute towards by sending us a pull request :) l

tayyars, 2019-10-25:
also :
 Lower transport layer has values in big endian byte order, not little endian

roshanrajaratnam, 2019-10-30:
@tayyars according the mesh specification 1.0.1 section 3.5.1 "All multiple-octet numeric values in this layer shall be sent in “big endian”, as described in Section 3.1.1.1" in the lower transport layer.

closed 2019-10-30 by roshanrajaratnam

tayyars, 2019-10-30:
> @tayyars according the mesh specification 1.0.1 section 3.5.1 "All multiple-octet numeric values in this layer shall be sent in “big endian”, as described in Section 3.1.1.1" in the lower transport layer.

 /**
     * Validates a unicast address
     *
     * @param address Address in bytes
     * @return true if the address is a valid unicast address or false otherwise
     */
    public static boolean isValidUnicastAddress(@NonNull final byte[] address) {
        if (!isAddressInRange(address)) {
            return false;
        }

        return isValidUnicastAddress(**MeshParserUtils.unsignedBytesToInt(address[0], address[1]**));
    }

 /**
     * Convert signed bytes to a 16-bit unsigned int.
     */
    public static int unsignedBytesToInt(byte b0, byte b1) {
        **return (unsignedByteToInt(b0) + (unsignedByteToInt(b1) << 8));**
    }

Should be ->
  **return (unsignedByteToInt(b0) << 8) + unsignedByteToInt(b1);** 

or i miss something ?

roshanrajaratnam, 2019-10-30:
The validation was wrong because of a missing `!` when checking if the address is in range. So in my humble opinion the current implementation is correct.

xiaofengming, 2019-12-06:
@roshanrajaratnam LowerTransportLayer.class 

------final int blockAckSrc = MeshParserUtils.unsignedBytesToInt(dst[1], dst[0]);----
and 
------MeshAddress.isValidUnicastAddress(dst)

return isValidUnicastAddress(**MeshParserUtils.unsignedBytesToInt(address[0], address[1]**));-----

There will always be a verification address

PR #265 (2019-10-30): Fix composition data get failure
Fixes a number of major issues.
* Fixes #258, #260 and this was happenning due to an incorrect validation of Unicast Address which is also fixed
* Fixes #257
* Fixes #262, the library does not use TAI anymore. 
* Fixes #254
PR #266 (2019-10-31): nRF Mesh Release v2.1.0
This release brings the following changes.
* Fixes #258, #260 and this was happenning due to an incorrect validation of Unicast Address which is
* Fixes #257
* Fixes #262, the library does not use TAI anymore. 
* Fixes #254 
* Fixes #245
* Fixes some UI inconsistencies in the group controls activity