nordicsemi/Android-nRF-Mesh-Library #258: Hi Warmly Team

theanhbp2018, 2019-10-18:
---
Hello
we can't provisioning with devices that have more than 2 elements, do you have plan in order to fix them ? 

thanks

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Platform details:**
 - Device: [e.g. Pixel XL]
 - OS: [e.g. Android 8.0]
 - Library Version (or Commit)

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster

roshanrajaratnam, 2019-10-18:
Hi which nRF Mesh SDK version are you referring to? Could you provide some more information please?

theanhbp2018, 2019-10-20:
applicationId "no.nordicsemi.android.nrfmeshprovisioner"
        minSdkVersion 18
        targetSdkVersion 29
        versionCode 53
        versionName "2.0.5"
        testInstrumentationRunner "androidx.test.runner.AndroidJUnitRunner"
        multiDexEnabled true
        vectorDrawables.useSupportLibrary = true

above is some info I used them,

roshanrajaratnam, 2019-10-20:
Hi, how about the nRF Mesh SDK version? Have you tested with some of the examples available in the sdk?

tayyars, 2019-10-21:
Hi ,
there a bug in isValidUnassignedAddress()
causing this problem.
#257

theanhbp2018, 2019-10-25:
> 
> 
> Hi, how about the nRF Mesh SDK version? Have you tested with some of the examples available in the sdk?

I used SDK 3.2.0.

theanhbp2018, 2019-10-25:
> Hi ,
> there a bug in isValidUnassignedAddress()
> causing this problem.
> #257

how I can fix them ?

roshanrajaratnam, 2019-10-25:
Hi, I'm currently out of office at the moment. I'll be back at work next week and I'll have a look at it. If you think 'isValidUnassignedAddress()' is causing the issue, check the comment from xiaofengming on #257 and try this

roshanrajaratnam, 2019-10-30:
@theanhbp2018 What's the nRF Mesh SDK version you are using on the firmware side?

roshanrajaratnam, 2019-10-30:
@theanhbp2018 Sorry for the delay I will take a look at this.

roshanrajaratnam, 2019-10-30:
@theanhbp2018 check out the fix on the following [branch](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/compare/bugfix/large-composition-data-get-failure)

closed 2019-10-30 by roshanrajaratnam

theanhbp2018, 2019-10-31:
thank you guys

PR #265 (2019-10-30): Fix composition data get failure
Fixes a number of major issues.
* Fixes #258, #260 and this was happenning due to an incorrect validation of Unicast Address which is also fixed
* Fixes #257
* Fixes #262, the library does not use TAI anymore. 
* Fixes #254
PR #266 (2019-10-31): nRF Mesh Release v2.1.0
This release brings the following changes.
* Fixes #258, #260 and this was happenning due to an incorrect validation of Unicast Address which is
* Fixes #257
* Fixes #262, the library does not use TAI anymore. 
* Fixes #254 
* Fixes #245
* Fixes some UI inconsistencies in the group controls activity