nordicsemi/Android-nRF-Mesh-Library #259: device not getting scanned on Android Version 10

shilpa108, 2019-10-18:
I have tried this app on Android version 9 and it is working fine but when tried on Android version 10, scanning does not show any device. Is there anything that I am missing or  I need to do something to support version 10? Please suggest

roshanrajaratnam, 2019-10-18:
Hi @shilpa, please disable/enable location permissions in the app settings and that should do the trick.

shilpa108, 2019-10-18:
Hi @roshanrajaratnam ,it is working now,Thanks.Is this the problem with Android 10 or what?

roshanrajaratnam, 2019-10-18:
@shilpa108 yes we did notice Android 10 seemed to have an issue with apps that already had location permission. After upgrading to Android 10 location permission settings were kind of lost eventhough the apps already had it.

closed 2019-10-18 by roshanrajaratnam