nordicsemi/Android-nRF-Mesh-Library #260: "COMPOSITION_DATA_GET_SENT",BLE no response

wanggan, 2019-10-21:
2019-10-21 16:15:38.330 30423-30423/com.anthero.mesh V/NetworkLayer: Encryption key: 92C7C87DEE55079074F72E9AC4099BF7
2019-10-21 16:15:38.330 30423-30423/com.anthero.mesh V/NetworkLayer: Privacy key: 5C8780F6B669DB0B9EA35EFF9850DACF
2019-10-21 16:15:38.331 30423-30423/com.anthero.mesh V/NetworkLayer: Sequence Number: 000019
2019-10-21 16:15:38.333 30423-30423/com.anthero.mesh V/NetworkLayer: Encrypted Network payload: 70D252C3A6290B7B62E55BB74D45
2019-10-21 16:15:38.354 30423-30423/com.anthero.mesh V/BleMeshManager: Sending data : 0x000F770849D04E2970D252C3A6290B7B62E55BB74D45
2019-10-21 16:15:38.357 30423-30423/com.anthero.mesh E/onMeshNodeStateUpdated: Sending composition data get...
2019-10-21 16:15:38.358 30423-30447/com.anthero.mesh V/BleMeshManager: Data written: 0x000F770849D04E2970D252C3A6290B7B62E55BB74D45
2019-10-21 16:15:38.358 30423-30447/com.anthero.mesh E/BleMeshManager: onCharacteristicWrite 00002add-0000-1000-8000-00805f9b34fb
2019-10-21 16:15:38.359 30423-30423/com.anthero.mesh I/ViewRootImpl: jank_removeInvalidNode all the node in jank list is out of time
2019-10-21 16:15:38.360 30423-30447/com.anthero.mesh V/MeshManagerApi: MeshNetwork pdu sent: 0x000F770849D04E2970D252C3A6290B7B62E55BB74D45

tayyars, 2019-10-21:
Hi , 
there a bug in isValidUnassignedAddress() 
causing this problem.
#257

roshanrajaratnam, 2019-10-22:
@tayyars I'm t currently out of office. I'll fix it as soon as I am back. Thanks for reporting.

shilpa108, 2019-10-30:
Hi ,
After changing  isValidUnassignedAddress() to as following as mentioned in #257 , still COMPOSITION_DATA_GET_SENT is not working
/**
* Validates an unassigned address
*
* @param address 16-bit address
* @return true if the address is a valid unassigned address or false otherwise
*/
public static boolean isValidUnassignedAddress(@nonnull final byte[] address) {
if (!isAddressInRange(address)) {
return false;
}
return isValidUnassignedAddress(MeshParserUtils.unsignedBytesToInt(address[0], address[1]));
}

roshanrajaratnam, 2019-10-30:
@shilpa108 Hi, I'm back at work. I'll take a look at this now. Could you confirm what nRF Mesh SDK version you are using?

shilpa108, 2019-10-30:
I am using  nRF Mesh SDK version 3.2.0

roshanrajaratnam, 2019-10-30:
@shilpa108 please checkout the [fix](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/compare/bugfix/large-composition-data-get-failure)

commit 19fa9e11 (2019-10-30): Fixes composition data get failure. This was happening due to an validating the unicast address incorrectly and due to this incorrect validation the segment acknowledgement timer was not starting.  Fixes #260, #258, #257
closed 2019-10-30 by roshanrajaratnam