nordicsemi/Android-nRF-Mesh-Library #263: When Send Transport Message to Node

LingoMeng, 2019-10-24:
Desc: When transport message bigger than 11 but smaller than 15, should send as unsegmented way. according to `Mesh Profile Spec 1.0.0 3.5.2.1 Unsegmented Access message.`
https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/b6d32cb97a2212f3dfe574ee093b93fafe7da752/android-nrf-mesh-library/meshprovisioner/src/main/java/no/nordicsemi/android/meshprovisioner/transport/LowerTransportLayer.java#L135-L149

```C
if (upperTransportPDU.length <= MAX_SEGMENTED_ACCESS_PAYLOAD_LENGTH) { 
should replace with:
if (upperTransportPDU.length <= MAX_UNSEGMENTED_ACCESS_PAYLOAD_LENGTH) { 
```

roshanrajaratnam, 2019-10-24:
@LingoMeng In my humble opinion the current implementation is correct. The lower transport layer only does the segmentation if the access message pdu is greater than 12 bytes. This is because the  max transport pdu can be 16 bytes including the 4 byte mic.

closed 2019-10-24 by roshanrajaratnam

LingoMeng, 2019-10-24:
While I can send max up to 11 octet access message in unsegmentation ways. According to mesh profile section 3.7.3 Access payload

LingoMeng, 2019-10-24:
> While I can send max up to 11 octet access message in unsegmentation ways. According to mesh profile section 3.7.3 Access payload

But in you currently code, when I send up to 9 octers access message, low transport layer will segmented, however 9octers access message able to send use unseg ways. According to mesh profile section 3.7.3 Access payload

roshanrajaratnam, 2019-10-25:
Ok I'll have to check this when I'm back at work. Don't really see why a 11 octet message would be sent as an unsegmented message and a 9 octet message is not.

roshanrajaratnam, 2019-10-31:
ok, however have you checked the transport mic size of your message?

roshanrajaratnam, 2019-10-31:
Do you have a fw I can try this issue?

LingoMeng, 2019-10-31:
`if (upperTransportPDU.length <= MAX_UNSEGMENTED_ACCESS_PAYLOAD_LENGTH) { `

You will know if you try it.  ￼

roshanrajaratnam, 2019-11-01:
ok will give ita go. I could probably use a GenericOnOffGet message to for a message shorter than 11 bytes.

Nuclear6, 2021-11-08:
hi, I want to understand the end result of this question, thanks!!!!