nordicsemi/Android-nRF-Mesh-Library #264: Auto connection for provisioned devices, Scene, Scheduler function, Auto bind App key, subcribe, public address..

theanhbp2018, 2019-10-24:
Hello Dear 
how do you think about that, It's very useful for us when moving around the Mesh area. and some function you can improve to help us,

thank you so much

roshanrajaratnam, 2019-10-30:
Hi @theanhbp2018 

This is a good feature to have but unfortunately this has to be implemented in a later stage. As of now after provisioning the provisioner connects automatically to the newly provisioned device, however does not reconnect automatically whena provisioned device in range. Since this is a sample app we have let the user decide what keys to bind, subscribe etc. Anyway we can look in to automatically connecting to a proxy when in range

theanhbp2018, 2019-10-31:
ok I got it , thanks but have you got any good idea to me ? ^^

roshanrajaratnam, 2019-11-06:
I would say, you will have to have the user pick, the Net Key, App Key, Groups and publication settings selected before starting the provisioning and then do the configuration right after the provisioning completes. Hope this helps.

closed 2019-11-06 by roshanrajaratnam