nordicsemi/Android-nRF-Mesh-Library #267: Editing network / app keys causes provisioning to halt

POHakansson, 2019-10-31:
Hi

I  have a large network created with 1.x versions of the app. I've now updated to v2.1.0 and I can't seem to provision new nodes anymore.

I've tried various methods for transferring my 1.x json file to v.2.x.x. For example I removed all the "-" in the UUIDs since there was a new check using PATTERN_UUID_HEX. This will allow me to at least open the JSON file. But I've noticed that there are a lot of small changes in the formatting.

Therefore I decided to try and start a new fresh network and see if I could later find some way of importing my old data once I have a new working "skeleton" JSON file.

However the end result is always that if I modify Network Settings to use my own keys, the provisioning sequence stops when the app tries to get the composition data. Maybe there is some place internally in the app where keys are not updated if they are edited in the Network Settings page?

It seems the problem is that the Get Composition Data PDU sent to the device can't be unencrypted properly.

**To Reproduce**
Steps to reproduce the behavior:
1. Create fresh Network Settings use "Reset"
2. Edit Network Key 1 and Application Key 1
3. Try provisioning a device
4. The sequence stops when trying to get the composition data.

**Platform details:**
 - Device: Galaxy S7 / Galaxy S10
 - OS: Android 8.0.0 / Android 9
 - Library Version v2.0.5 and v.2.1.0 7790b170679116e98d built w Android Studio

roshanrajaratnam, 2019-10-31:
@POHakansson Thanks for reporting this. It is reproduceable and i'll try to get a fix out

EnMMSegur, 2019-11-07:
I think I found the problem, during the provisioning the unicast address of all the nodes is set to 1.

roshanrajaratnam, 2020-05-13:
@POHakansson Sorry for the delayed reply. Unfortunately I have been busy with a number of other tasks. Finally I managed to figure out the issue here. This has nothing to do with the Unicast Address or the Composition Data PDU or the provisioning. The provisioning succeeds but the initial configuration fails. The reason is that when creating a `NetworkKey` an `IdentityKey `is generated for each key. However editing a `NetworkKey `does not update the `Identitykey`. This causes the device to be not found right after the provisioning is completed because the device advertises with a Node Identity and this node identity is derived using the above mentioned `IdentityKey`.

POHakansson, 2020-05-13:
Excellent, thanks for fixing it!

closed 2020-06-15 by roshanrajaratnam