nordicsemi/Android-nRF-Mesh-Library #268: Provisioning process is corrupted after the last update (2.1.0)

mtiutiu, 2019-11-01:
---

**Describe the bug**
After hours of scratching my head and almost pulling my hair out I found out that the new release of this great application screws things up on my **Apache Nimble** powered mesh nodes when trying to provision a new one.

**To Reproduce**
Steps to reproduce the behavior:
1. Provision a new node via the app
2. Click on Identify Node (**OOB** is **disabled**)
3. Wait until the node gets provisioned and the nrfMesh app reconnects after it completes
4. During the reconnection phase when it tries to read the new provisioned node and to get its composition data the nrfMesh app hangs at the **Sending block acknowledgements** step
5. Then I have to press the back button from my phone to get back to the main screen where I can see my node in the list as being already provisioned
6. Then I try to connect to it and seems to work but I get this screen:
```
Transaction failed!
All message segments were not received, operation timed out!
```
7. Then by pressing OK I get to the node main screen (in the upper left corner I see that it's connected) but the Elements (Composition Data) list is empty

**Expected behavior**
The provisioning should work from start to end and the connection to the newly provisioned node should happen automatically.
The provisioning complete message window should appear.

**Platform details:**
 - Device: **Xiaomi Mi A2 Lite**
 - OS: **Android One**
 - Library Version: **2.1.0**

**Logs / Screenshots**
This is the log from my custom **Apache Nimble** based mesh node:

```
000000 Initializing...
000000 Bluetooth initialized
000049 Mesh initialized
002112 provisioning complete for net_idx 0x0000 addr 0x0002
003391 [ts=26492156us, mod=4 level=3] seg_tx_send_unacked: Ran out of retransmit attempts
...
```
So even if the above occurs the node gets provisioned BUT I need to re-connect to it and the Elements (Composition data) area is empty - I can issue a Get Composition Data and it fetches it. Still the provisioning process is corrupt and incomplete due to the above errors.
Sorry I cannot provide more details for now as I'm just learning **Apache Nimble** and mesh networking internals in general.
What I can say is that if I revert back to the previous version of the nrfMesh app (**2.0.5**) then everything works smooth just as before.

Thanks for this great app.

roshanrajaratnam, 2019-11-01:
Hi, thanks for your feedback and sorry for any inconvenience caused. Could you kindly share a hex file so that I can test this first thing on Monday morning?

mtiutiu, 2019-11-04:
@roshanrajaratnam Hi, I can do that not a problem but I don't know how useful it will be for you as it doesn't work without a bootloader (Apache Mynewt uses `mcuboot` to load the main application during runtime although I don't think it's quite mandatory but I'm not sure yet because I wasn't able to run anything without it). 

So instead of sending you the hex file you can for example compile my example (derived from their official examples) as described [here](https://github.com/mtiutiu/Coding_Playground/tree/master/sensors_network/project/LivoloLightsNode/newt). You need to clone my git repo of course.

In that example I'm using `nordic_pca10040` but you can use any other supported bsp from [here](https://github.com/apache/mynewt-core/tree/master/hw/bsp). Some of them support flashing/testing by using a `JLink` probe (pcaxxx boards), others via `OpenOCD` and a `DAPLink/STLink` adapter.

Hope it helps.

roshanrajaratnam, 2019-11-18:
@mtiutiu Sorry for the delayed response. the latest release fixes this issue.

closed 2019-11-18 by roshanrajaratnam