nordicsemi/Android-nRF-Mesh-Library #270: The second time Provisioning, APP uses 'Proxy Service' for Provisioning

BleonDev, 2019-11-09:
After the first Provisioning succeeds; Then, Reset Node and start Provisioning for the second time. APP uses   "Proxy Service"  for Provisioning. At this point, we need to restart bluetooth for Provisioning. 
  
**Logs / Screenshots**
**step1**:After Provisioning succeeds for the first time, we will Reset Node in APP.
[image]

**step2**:Then a second Provisioning, when the app is Provisioning with Proxy Service instead of Proxy Service 
[image]；Then there is provisioning Service on the device, not proxy Servce  
[image]

**step2**: We need to reopen bluetooth for Provisioning
[image]

Thank you.

philips77, 2019-11-09:
It's a caching issue. The library forces clearing the cache after provisioning, but not after resetting. It's a room for improvement.

BleonDev, 2019-11-11:
> It's a caching issue. The library forces clearing the cache after provisioning, but not after resetting. It's a room for improvement.

invoke refreshDeviceCache after provisioning,and refreshDeviceCache return is true; but next provisioning still use "proxy service"
[image]

closed 2019-11-18 by roshanrajaratnam