nordicsemi/Android-nRF-Mesh-Library #280: Automatic Connection

theanhbp2018, 2019-11-24:
I saw in iOS App had been new function can able to connect by Proxy, I thought should do that in android App. You can think more.

thanks

roshanrajaratnam, 2019-12-02:
Do you mean automatically connect to proxies in range? Yes this is something that can be added.

theanhbp2018, 2019-12-04:
I am here waiting for you

roshanrajaratnam, 2019-12-04:
Hi @theanhbp2018 unfortunately this is not something we will do right now due to some other priorities but will make sure to add this down the line.

DevHachi, 2021-06-15:
I wanted to ask. If there is any progress to that. Cause I would also like to have this functionality this would reduce significally the amount of work to do this myself.

roshanrajaratnam, 2021-06-16:
Hi, we have not done any work on this and will have to wait due to some other priorities.

closed 2023-05-25 by roshanrajaratnam

Fabbros, 2024-01-31:
@roshanrajaratnam I'm also interested in this point. I understand that this is scheduled sometime soon, but can you get me a pointer to something useful to build something custom by myself?

roshanrajaratnam, 2024-07-19:
Hi, I'm out of office at the moment, however you could take a look at the scanning and connection functionality  ScannerRepository class. You could write your own function based on this to connect automatically on app startup. Hope this helps.