nordicsemi/Android-nRF-Mesh-Library #282: Generic Level's Move: Remaining time issue

vikrant8051, 2019-11-30:
When Gen. Level's Move is in progress & Client sends level get command
then even when NODE send Remaining time as 0x3F in response, nRFMesh 
App shows Remaining Time as 6300 ms.

In this case, Rem Time should be display as "UNKWOWN"

roshanrajaratnam, 2019-12-02:
I think this might need to be reported [here](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library) as currently Android version does not support Generic Level Move.

closed 2019-12-02 by roshanrajaratnam

vikrant8051, 2019-12-02:
Even though Move (which is related to Gen. Level state) is not supported kindly note that
Gen. Level could bounded with any other states (like Gen. OnOff, Light Lightness , Light CTL)

In that case, when client will ask status (when MOVE is in progress) via 
 - gen. onoff get or
- gen, level get
- light lightness actual get or
- light ctl get
then here remaining time should get as "Unknown"