nordicsemi/Android-nRF-Mesh-Library #283: Publishing to all nodes

deverickasih, 2019-12-04:
Hi, i am an iOS developer. I'm asking on behalf of an android developer.

In the iOS version of this library, i am capable of setting an element-model to publish to all nodes and when i touch the physical button of the element-model, in the iOS app, it instantly detect the changes of 'on' & 'off'. We tried to do the exact same thing in the android app, but in the app, the on/off state does not change. In other words, the android app with regards to 'publishing to all nodes' function either does not work the same way as the iOS version or there is a bug that currently has not been fixed.

Please help.

roshanrajaratnam, 2019-12-04:
Hi @deverickasih this is because the sample iOS app automatically adds all nodes address to proxy whitelist. However in the Android version you have to do this manually. This should do the trick I believe. Let me know if it works for you.

deverickasih, 2019-12-05:
We have recently tried and found out that if we set blacklist on the sample app and add no addresses, we are capable of receiving the states of any of the element-models we have in the mesh network.

But if we choose whitelist and add the addresses we want to receive states from, when we toggle on/off on the element-models we have whitelisted, we do not get state updates from them.

Is this the intended workings of the sample app?

philips77, 2019-12-05:
Can you confirm, that you get the same results using the iOS nRF Mesh app? If yes, there's a bug in yours proxy firmware, or is doesn't fully support proxy filter. If not, and iOS app behaves correctly, is an Android app issue.

deverickasih, 2019-12-06:
We have two boards (A & B) using example firmware: 
light_switch_server_nrf52832_xxAA_s132_6.1.1_merged.hex

We also set the on/off generic model for each board to publish to all nodes (0xFFFF)

Case 1 : Use Android proxy filter and set whitelist with address 0xFFFF
When we toggle on/off on the physical board A, in the app we receive updates for A
But without changing the view in the app (staying inside the view for A), when we toggle on/off on the physical board B, in the app we also receive updates

Case 2 : Use Android proxy filter and set blacklist without setting addresses
The same thing happens like Case 1

In iOS nRF Mesh, it works fine. No cross board problem. When we toggle on/off on the physical board B, the iOS app does not show any updates for board A.

roshanrajaratnam, 2020-05-14:
@deverickasih sorry for the delayed reply due to being busy with other tasks. I found the issue relating to the list of addresses not showing up. However not entirely sure how to reproduce the issue here. Could you share a more detailed description to how you have setup the boards so that we can run through the exact same steps to reproduce.

roshanrajaratnam, 2020-06-04:
Closing due to inactivity, but please re-open if you think the problem persists and describe step by step how to reproduce this.

closed 2020-06-04 by roshanrajaratnam