nordicsemi/Android-nRF-Mesh-Library #284: Lists only two provisioned devices in Network tab

hardiksingh-rathore, 2019-12-06:
---

**Describe the bug**
- If I provision three mesh nodes (A,B and C), the app list only the last two devices(it's A & C OR B & C)

**To Reproduce**
Steps to reproduce the behavior:
1. Try provisioning  3 - 4 devices

**Expected behavior**
Should list all provisioned mesh nodes

**Platform details:**
 - Device: Galaxy A10s
 - OS: Android 9.0
 - App Version - 2.1.4

**Logs / Screenshots**
[image]

In screen shot 0x0003 and 0x0005 are visible and 0x0004 is missing. (it was visible before provisioning 0x0005)

roshanrajaratnam, 2019-12-06:
@hardiksingh-rathore do you know if the devices have unique device uuids? I suppose the device got overwritten because the device UUIDs were the same?

hardiksingh-rathore, 2019-12-12:
Yes. the problem was because of the same UUIDs.
Thanks!

closed 2019-12-12 by hardiksingh-rathore