nordicsemi/Android-nRF-Mesh-Library #289: Is BLE5 supported in this app

MeenalBahua, 2020-01-27:
I was wondering if the current android app supports Ble5 ?

philips77, 2020-01-27:
What Ble 5 features do you have in mind? Bluetooth Mesh is designed to work with devices supporting Bluetooth 4.0+.

MeenalBahua, 2020-01-28:
We are looking for BLE5 long range mode for 125kbps speed.

philips77, 2020-01-28:
The current spec does not mention BLE5 features. The ADV Bearer by spec should use PHY LE 1M, same as the proxy connection. But for private single vendor installations, as far as I know, nRF5 Mesh can be tweaked to work using 125kbps PHY. It is not implemented in Nordic Mesh SDK, however.
Now, when modern Android phones support Long Range, it is possible to experiment with phones as well, but this is not in our focus at the moment.

roshanrajaratnam, 2020-03-05:
Closing this issue for now.

closed 2020-03-05 by roshanrajaratnam