nordicsemi/Android-nRF-Mesh-Library #290: Does it only support Nordic devices?

SonnySwolf, 2020-02-29:
---

**Describe the bug**
the android app can only provsion nordic device. and aroha and sillicon devices can't complete the provision. but ios app works fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'ADD NODE'
2. Click on 'device'
3. Click on 'IDENTIFY'
3. app send 03 00 05 to the device, but the device not reply.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Platform details:**
 - Device: [HUAWEI Mate 10 Pro]
 - OS: [e.g. Android 9.1]
 - Library Version 2.1.4

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster
[provision abort for AB1611.txt](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/4270139/provision.abort.for.AB1611.txt)

roshanrajaratnam, 2020-03-02:
Hi  does the device identify itself when you send this message? for example blink?

SonnySwolf, 2020-03-04:
Hi ,
    I can get AB1611 log, AB1611 received 03 00 05 , but it not reply and report a error. provisiong data should be ignored. 
[image]
[android无法入网.log](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/4284783/android.log)

roshanrajaratnam, 2020-03-04:
Hi @SonnySwolf not entirely sure why this device is ignoring the provisioning data. What's this device AB1611?

SonnySwolf, 2020-03-04:
Hi ,
    sorry, AB1611 not nordic's bt chip,  it is airoha's bt chip. I also using silicon lab's bt chip do same test. silicon lab's bt chip with same problem.

roshanrajaratnam, 2020-04-16:
Hi @SonnySwolf have you had any luck with your issue?

SonnySwolf, 2020-04-16:
yes, this is my device fault.

closed 2020-04-16 by SonnySwolf