nordicsemi/Android-nRF-Mesh-Library #292: Notification from the Mesh Network

clma91, 2020-03-04:
**Is your feature request related to a problem? Please describe.**
Do I understand correctly that it is currently not possible to receive messages via the API mesh?

**Describe the solution you'd like**
It would be very helpful if there was an API function/callback to register on mesh messages to receive status messages from nodes.

**Describe alternatives you've considered**
Is it correct to assume that the alternative would be for the nodes to send corresponding BLE notifications?

**Additional context**
We can send Mesh Messages through the SDK via the MeshManagerApi createMeshPdu with the GroupAddress and the Message but we can not receive any Mesh Messages sent from a device.

roshanrajaratnam, 2020-03-05:
@clma91 You should be able to receive the corresponding status message from the device to any of the messages sent vfrom the app/library. For example if the phone sends an acknowledged generic on off the app should receive the status and this applies to the configuration messages as well.

clma91, 2020-03-05:
Alright, thank you very much for your answer. But the problem was actually not in our react native app but rather the firmware, developed by an other departement. This firmware version was not sending these messages over GATT. 
But now I do not get the messages by any [MeshStatusCallbacks](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/4b29037c92cd83a75f0a1ce65c7be528e99c5142/android-nrf-mesh-library/meshprovisioner/src/main/java/no/nordicsemi/android/meshprovisioner/MeshStatusCallbacks.java#L95) but through [handleNotifications()](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/b6d32cb97a2212f3dfe574ee093b93fafe7da752/android-nrf-mesh-library/meshprovisioner/src/main/java/no/nordicsemi/android/meshprovisioner/MeshMngrApi.java#L49). I may also say that I just reveived the firmware update and had just a quick look into this. Maybe I could catch these messages also via callbacks.

To your question:
We send mesh messages via [MeshManagerApi.createMeshPdu()](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/4b29037c92cd83a75f0a1ce65c7be528e99c5142/android-nrf-mesh-library/meshprovisioner/src/main/java/no/nordicsemi/android/meshprovisioner/MeshManagerApi.java#L760) by specifying a [VendorModelMessageUnacked ](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/87398b39004855f9e8eb54b840225104c3f1bc3c/android-nrf-mesh-library/meshprovisioner/src/main/java/no/nordicsemi/android/meshprovisioner/transport/VendorModelMessageUnacked.java#L13) and a corresponding group address.
I'm not sure if this is the way to go, but on the firmware-side the guys receive this message and can process it.

Thanks for the hint with the acknowledge message - we can definitely use them too.

closed 2020-03-05 by clma91

roshanrajaratnam, 2020-03-05:
You stil don't get the MeshStatusCallbacks? Have you called `setMeshStatusCallbacks()`, check [here](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/b5125246d0fd6af225a272990b01b8f1a65050e6/Example/nrf-mesh/app/src/main/java/no/nordicsemi/android/nrfmeshprovisioner/viewmodels/NrfMeshRepository.java#L165). The library will parse the raw notification bytes and give you a MeshMessage.

clma91, 2020-03-05:
No i don't, but like I said the problem now is not on the react native app but rather on the firmware. The network header is not yet filled correctly. As soon as this comes back correctly to the app we should also get the callback. 

Thank you for your help.