nordicsemi/Android-nRF-Mesh-Library #293: Access PDU Op codes computed differently in iOS implementation

kiyoshi-brilliant, 2020-03-24:
---

**Describe the bug**
Logic to compute Op codes does not match iOS implementation. Also filing on iOS repo.
Most concerning is the default / else case treatment but the other cases are slightly mismatched as well.
Please provide guidance as to which is correct.

Android:
https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/fd90df15ebb03222ee6f7f7bb52587e1f7e60173/android-nrf-mesh-library/meshprovisioner/src/main/java/no/nordicsemi/android/meshprovisioner/utils/MeshParserUtils.java#L357

iOS:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/68dc144aabdf99da472b809115d94358b843343d/nRFMeshProvision/Classes/Layers/Access%20Layer/AccessPdu.swift#L151

**To Reproduce**
Steps to reproduce the behavior:
1. View Op code computing code

**Expected behavior**
Should be logically equivalent.

**Platform details:**
N/A

**Logs / Screenshots**
N/A

roshanrajaratnam, 2020-04-16:
Well you are correct they do look different but they both have the same function as intended. However I have made it a little bit similar.

closed 2020-04-16 by roshanrajaratnam

kiyoshi-brilliant, 2020-04-16:
Thanks! I think we had gotten confused about passing the op code w/ vs w/o the company identifier, we appreciate the changes!

PR #297 (2020-04-16): Improve opcode access layer opcode parsing
* Fixes #293 - Improvements to how the Vendor model opcode is constructed. Now the ser may or or may not enter the opcode in the UI with the 3-octet opcode bits set. This is to make both implementations look similar to a certain extent.
* Fixes #287  by improving how the opcode is being parsed.
PR #302 (2020-04-30): Release 2.2.0
* Removes the duplicate sequence number field in the provisioner table. 
  - If needed to change the sequence number of a provisioner, select the provisioner node from the nodes list using the `provisionerUuid` and use `node.setSequenceNumber(sequenceNumber)`
* * Fixes #293 - Improvements to how the Vendor model opcode is constructed. Now the ser may or or may not enter the opcode in the UI with the 3-octet opcode bits set. This is to make both implementations look similar to a certain extent.
* Fixes #287  by improving how the opcode is being parsed.
* Fixes #301
* Fixes #244 IV [... cut]