nordicsemi/Android-nRF-Mesh-Library #294: Reset node crash App

dqapollo2018, 2020-04-03:
Hello Dear 
I have seen issue as tittle, please tell me know how to fix them

many thanks

**this issue**

E/AndroidRuntime: FATAL EXCEPTION: AsyncTask #2
    Process: com.dienquang.android.motobike, PID: 29309
    java.lang.RuntimeException: An error occurred while executing doInBackground()
        at android.os.AsyncTask$3.done(AsyncTask.java:354)
        at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
        at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
        at java.util.concurrent.FutureTask.run(FutureTask.java:271)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:764)
     Caused by: java.util.ConcurrentModificationException
        at java.util.ArrayList$Itr.next(ArrayList.java:860)
        at java.util.Collections$UnmodifiableCollection$1.next(Collections.java:1084)
        at androidx.room.EntityDeletionOrUpdateAdapter.handleMultiple(EntityDeletionOrUpdateAdapter.java:86)
        at no.nordicsemi.android.meshprovisioner.data.ProvisionedMeshNodesDao_Impl.update(ProvisionedMeshNodesDao_Impl.java:315)
        at no.nordicsemi.android.meshprovisioner.MeshNetworkDb$UpdateNetworkAsyncTask1.doInBackground(MeshNetworkDb.java:417)
        at no.nordicsemi.android.meshprovisioner.MeshNetworkDb$UpdateNetworkAsyncTask1.doInBackground(MeshNetworkDb.java:384)
        at android.os.AsyncTask$2.call(AsyncTask.java:333)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
        at java.lang.Thread.run(Thread.java:764) 
---

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Platform details:**
 - Device: [e.g. Pixel XL]
 - OS: [e.g. Android 8.0]
 - Library Version (or Commit)

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster

roshanrajaratnam, 2020-04-03:
Are you trying reset multiple nodes at the same time?

dqapollo2018, 2020-04-04:
> Are you trying reset multiple nodes at the same time?
uhm, that issue sometimes is occur.

dqapollo2018, 2020-04-10:
can you help me ?

roshanrajaratnam, 2020-04-14:
Hi @dqapollo2018 each node that is being reset is deleted from the network so if you are iterating through the same list of nodes you are bound to get ConcurrentModificationException because the same list is getting modified. I would also suggest to when you reset nodes, wait till you get the response before continuing with the next node.

roshanrajaratnam, 2020-04-22:
I'll close this issue and please feel free to re-open.

closed 2020-04-22 by roshanrajaratnam