nordicsemi/Android-nRF-Mesh-Library #298: check isProvisioningComplete

benjak135765, 2020-04-18:
Hi~
How can i get the "registerApp() - UUID" ?

I want check the device 
if not exist in nodes then isProvisioningComplete = false
at red area 

 
[image]

roshanrajaratnam, 2020-04-20:
@benjak135765 this is an Android system log. What are you trying to achieve?

benjak135765, 2020-04-20:
my app want scan Provisioned Device.
only connect it, not Provisioning again
but i don'w know how can i get the node...

roshanrajaratnam, 2020-04-21:
@benjak135765 I guess you mean that you want to  connect to the same device after restarting the app? You will have to scan for devices with devices advertising with the following 16-bit UUID 0x1828 and also match against the network id generated using the network key. Please refer to the sample app.

closed 2020-05-14 by roshanrajaratnam