nordicsemi/Android-nRF-Mesh-Library #306: unable to provision the node on esp32 using the nrf version 2.2.0

ankit-thealchemist, 2020-05-12:
Hello, 
**Describe the bug**
while provisioning the node using the example [https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server/main](url) , on esp32 dev kit v1, I am getting the error says prohibited.  nrf mesh version (2.2.0)

while I downloaded the old version of nrf mesh 2.1.4, it works flawlessly

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
[image]

**Expected behavior**
The current version of the nrf mesh should  provisioned the node as the old nrf mesh version 2.1.4.

**Platform details:**
 - Device: Redmi note 4 / redmi note 7 
 - OS: Android 6.0 & 8.0
 - Library Version (or Commit)

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster

roshanrajaratnam, 2020-05-12:
@ankit-thealchemistT that's strange that it's not working. It should work on any device and we have no issues with our mesh sdk examples. Could you provide me with a compiled hex file I can flash on a 52840 device please?

zacwbond, 2020-05-20:
@ankit-thealchemist @roshanrajaratnam 
FYI, I am experiencing this same problem, a "prohibited" message when I try to provision an ESP32 device that is running their esp_ble_mesh.

Interestingly, I am able to provision an ESP32 perfectly well if I use the Nimble mesh implementation; it only fails with the esp_ble_mesh implementation. 

The error occurs during the public key exchange.  Here's some debug.  You can see at the end that the ESP32 generates a public key, then swaps X and Y and changes to big endian before sending it split across four notifies.  This is the point at which I get the "prohibited" message.  As soon as I tap the message, the app disconnects from the ESP32.  

I have not yet looked at the app code to see what causes that message.

```
I (23603) MESH: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
connection updated; status=0
D (28352) BLE_MESH: proxy_char_access_cb, write: handle = 18, len = 3, data = 030005
D (28352) BLE_MESH: Mesh Provisioning PDU
D (28352) BLE_MESH: ==== RX 2 bytes: 0005
W (28353) BLE_MESH: ==== rx type 002x len 1
W (28353) BLE_MESH: ===send invite
D (28354) BLE_MESH: Attention Duration: 5 seconds
W (28354) BLE_MESH: bt_mesh_attention, No Health Server context provided
D (28355) BLE_MESH: SEND conn 0x3ffc6a04 type 0x03 len 12: 010100010000010005000000
D (28356) BLE_MESH: 13 bytes: 03010100010000010005000000
GATT procedure initiated: notify; att_handle=20
E (31002) Heap: 53768
E (31002) HeapMin: 52832
D (39613) BLE_MESH: proxy_char_access_cb, write: handle = 18, len = 7, data = 03020000020001
D (39614) BLE_MESH: Mesh Provisioning PDU
D (39614) BLE_MESH: ==== RX 6 bytes: 020000020001
W (39614) BLE_MESH: ==== rx type 202x len 5
I (39615) BLE_MESH: Algorithm:   0x00
I (39615) BLE_MESH: Public Key:  0x00
I (39616) BLE_MESH: Auth Method: 0x02
I (39616) BLE_MESH: Auth Action: 0x00
I (39617) BLE_MESH: Auth Size:   0x01
D (39618) BLE_MESH: bt_mesh_rand, rand: 9006f4a9
D (39618) BLE_MESH: btc_ble_mesh_output_number_cb
D (39619) BLE_MESH: btc_ble_mesh_prov_callback
I (39619) MESH: ESP_BLE_MESH_NODE_PROV_OUTPUT_NUMBER_EVT: number is 8
D (39662) BLE_MESH: proxy_char_access_cb, write: handle = 18, len = 20, data = 4303e2247b972024e43085ea9b5c71c98a8c6f28
D (39711) BLE_MESH: proxy_char_access_cb, write: handle = 18, len = 20, data = 836f9f6a393f45a40ecae4b6537634db3d6cbf43
D (39712) BLE_MESH: proxy_char_access_cb, write: handle = 18, len = 20, data = 8386bb74582fac022882ac260944625699bb4950
D (39713) BLE_MESH: proxy_char_access_cb, write: handle = 18, len = 9, data = c31cfc0886d3525261
D (39714) BLE_MESH: Mesh Provisioning PDU
D (39714) BLE_MESH: ==== RX 65 bytes: 03e2247b972024e43085ea9b5c71c98a8c6f286f9f6a393f45a40ecae4b6537634db3d6cbf4386bb74582fac022882ac260944625699bb49501cfc0886d35252
W (39716) BLE_MESH: ==== rx type 302x len 64
D (39717) BLE_MESH: Remote Public Key: e2247b972024e43085ea9b5c71c98a8c6f286f9f6a393f45a40ecae4b6537634db3d6cbf4386bb74582fac022882ac260944625699bb49501cfc0886d3525261
W (39719) BLE_MESH: ===send public key
D (39719) BLE_MESH: private key = bd1a3ccda6b8995899b740eb7b60ff4a503f10d2e3b3c974385fc5a3d4f6493f
D (40083) BLE_MESH: 0x3ffe573c
D (40083) BLE_MESH: DHkey: 9efa88678c1d1abb0441dee912707118321409c3c365bc6f11416f7d4a77d061

D (40083) BLE_MESH: Local Public Key: e69d350e480103ccdbfdf4ac1191f4efb9a5f9e9a7832c5e2cbe97f2d203b0208bd28915d08e1c742430ed8fc24563765c15525abf9a32636deb2a65499c80dc

D (40085) BLE_MESH: SEND conn 0x3ffc6a04 type 0x03 len 65: 0320b003d2f297be2c5e2c83a7e9f9a5b9eff49111acf4fddbcc0301480e359de6dc809c49652aeb6d63329abf5a52155c766345c28fed3024741c8ed01589d2

D (40087) BLE_MESH: 20 bytes: 430320b003d2f297be2c5e2c83a7e9f9a5b9eff4
GATT procedure initiated: notify; att_handle=20
D (40089) BLE_MESH: 20 bytes: 839111acf4fddbcc0301480e359de6dc809c4965
GATT procedure initiated: notify; att_handle=20
D (40090) BLE_MESH: 20 bytes: 832aeb6d63329abf5a52155c766345c28fed3024
GATT procedure initiated: notify; att_handle=20
D (40092) BLE_MESH: 9 bytes: c3741c8ed01589d28b
GATT procedure initiated: notify; att_handle=20

E (41002) Heap: 53660
E (41002) HeapMin: 50860
E (51002) Heap: 53660
E (51002) HeapMin: 50860
subscribe event; conn_handle=0 attr_handle=20 reason=2 prevn=1 curn=0 previ=0 curi=0
D (58967) BLE_MESH: len 2: 0000
W (58967) BLE_MESH: Client wrote 0x0000 instead enabling notify
disconnect; reason=531
```

roshanrajaratnam, 2020-05-22:
Hi @zacwbond  thanks for the detailed feedback. Could you try the same with our nRF Mesh iOS app as well to see if you could reproduce the same?

Campou, 2020-05-26:
Hi @ankit-thealchemist @roshanrajaratnam @zacwbond 

Have found the root cause, because esp32 uses 23 as the default MTU size, which causes the failure on the App side. After changing the MTU size to 69, the provisioning procedure is completed successfully.

I have tried with the nRF Mesh Android App (version 2.1.3) and the latest iOS App, this issue doesn't exist.

Thanks.

roshanrajaratnam, 2020-05-26:
@Campou yes I had a feeling this could be the MTU and yes thanks for confirming this. So the app does a default MTU request and can you check the logs if you get a response for this? If the app does not get a response it uses the default MTU which is 23 and if you look at the logs everything is sent in chunks of 23.

ankit-thealchemist, 2020-05-27:
> Hi @ankit-thealchemist @roshanrajaratnam @zacwbond
>
> Have found the root cause, because esp32 uses 23 as the default MTU size, which causes the failure on the App side. After changing the MTU size to [...]

yes changing the mtu to the 69 helps. It just connect to the nrf mesh but does connect to the other apps like RSL 10 mesh by on semiconductor.

ankit-thealchemist, 2020-05-27:
> 43
@roshanrajaratnam @zacwbond @Campou The most unusual thing is that when mtu is set at default 23 and the esp 32 sends the provisioning PDU, of length of 65 bytes, 
D (40085) BLE_MESH: SEND conn 0x3ffc6a04 type 0x03 len 65: 0320b003d2f297be2c5e2c83a7e9f9a5b9eff49111acf4fddbcc0301480e359de6dc809c49652aeb6d63329abf5a52155c766345c28fed3024741c8ed01589d2

it split into the four packet according to the mtu. check below

> D (40087) BLE_MESH: 20 bytes: 430320b003d2f297be2c5e2c83a7e9f9a5b9eff4
GATT procedure initiated: notify; att_handle=20
D (40089) BLE_MESH: 20 bytes: 839111acf4fddbcc0301480e359de6dc809c4965
GATT procedure initiated: notify; att_handle=20
D (40090) BLE_MESH: 20 bytes: 832aeb6d63329abf5a52155c766345c28fed3024
GATT procedure initiated: notify; att_handle=20
D (40092) BLE_MESH: 9 bytes: c3741c8ed01589d28b
GATT procedure initiated: notify; att_handle=20

The thing if mtu is set such as the segmentation of the packet is limited to 2(such as mtu size 37), the nrf mesh provisioned the device. It seems that nrf mesh app is not processing the pdu packet if it is segmented in more than 2 packet. 

Like that I did, set the mtu size to 36

D (28126) BLE_MESH: conn 0x3ffc7cc8 type 0x03 len 65: 03ff7419ba63a3523a8db71c803cea96a0d8525c0aa8ee17c498c911510cec8ae41efd827d0b9dbb35a542ace3e7208ba81b5c5d9e6a154e6176835a54757494
D (28136) BLE_MESH: 34 bytes: 4303ff7419ba63a3523a8db71c803cea96a0d8525c0aa8ee17c498c911510cec8ae4
D (28146) BLE_MESH: 33 bytes: c31efd827d0b9dbb35a542ace3e7208ba81b5c5d9e6a154e6176835a54757494a3
D (28286) BLE_MESH: bt_mesh_bta_gatts_cb, write: handle = 48, len = 18, data = 0305f3549f5fec82b3869292e8564a82441b
D (28296) BLE_MESH: Mesh Provisioning PDU
D (28296) BLE_MESH: 17 bytes: 05f3549f5fec82b3869292e8564a82441b
D (28296) BLE_MESH: Remote Confirm: f3549f5fec82b3869292e8564a82441b
D (28306) BLE_MESH: ConfInputs[0]   050300010000000000000000000000000014bfdf8a2fcd5764ffe5acd798447a2aec03bc879122104df1a3febaf1993500d615856016073865c95ff3b77f36ff
D (28316) BLE_MESH: ConfInputs[64]  7117beefba6e7525538dfb5b68eb2dd637ff7419ba63a3523a8db71c803cea96a0d8525c0aa8ee17c498c911510cec8ae41efd827d0b9dbb35a542ace3e7208b
D (28336) BLE_MESH: ConfInputs[128] a81b5c5d9e6a154e6176835a54757494a3
D (28346) BLE_MESH: ConfirmationSalt: a0c7a465e07983ac8815214cc61324b2
D (28346) BLE_MESH: ConfirmationKey: 874c95e05ea475a22aee5c57161a3d67
D (28356) BLE_MESH: bt_mesh_rand, rand: 389ae503796298e76a6924150f535742
D (28366) BLE_MESH: LocalRandom: 389ae503796298e76a6924150f535742
D (28366) BLE_MESH: ConfirmationKey 874c95e05ea475a22aee5c57161a3d67
D (28376) BLE_MESH: RandomDevice 389ae503796298e76a6924150f535742
D (28376) BLE_MESH: AuthValue 00000000000000000000000000000000
D (28386) BLE_MESH: conn 0x3ffc7cc8 type 0x03 len 17: 0545cade1d38b19a99aeb4bb0835feb87e
D (28396) BLE_MESH: 18 bytes: 030545cade1d38b19a99aeb4bb0835feb87e
D (28486) BLE_MESH: bt_mesh_bta_gatts_cb, write: handle = 48, len = 18, data = 03061e67a2fc3013b045ae3cf9ba72cb15db
D (28486) BLE_MESH: Mesh Provisioning PDU
D (28486) BLE_MESH: 17 bytes: 061e67a2fc3013b045ae3cf9ba72cb15db
D (28496) BLE_MESH: Remote Random: 1e67a2fc3013b045ae3cf9ba72cb15db
D (28506) BLE_MESH: ConfirmationKey 874c95e05ea475a22aee5c57161a3d67
D (28506) BLE_MESH: RandomDevice 1e67a2fc3013b045ae3cf9ba72cb15db
D (28516) BLE_MESH: AuthValue 00000000000000000000000000000000
D (28526) BLE_MESH: conn 0x3ffc7cc8 type 0x03 len 17: 06389ae503796298e76a6924150f535742
D (28526) BLE_MESH: 18 bytes: 0306389ae503796298e76a6924150f535742
D (28536) BLE_MESH: ProvisioningSalt: 25d6c9c7549e1bcb42414bcc9f1c2e6a

roshanrajaratnam, 2020-05-27:
@ankit-thealchemist Thanks for the detailed reply, yes you are correct the app should handle the number of packets dynamically as far as I remember I have not seen this issue happenning with lower MTU with the Nordic Mesh SDK. Not sure what caused this exactly, i'll do some debugging and get back to you, but unfortunately it might have to wait till Monday.

Thecave3, 2020-05-29:
> Hi @ankit-thealchemist @roshanrajaratnam @zacwbond
>
> Have found the root cause, because esp32 uses 23 as the default MTU size, which causes the failure on the App side. After changing the MTU size to [...]

I'm having the same issue and I'm trying to change the MTU of the esp while this is going to be solved. Can you explain how you've done that?

ankit-thealchemist, 2020-05-30:
> I'm having the same issue and I'm trying to change the MTU of the esp while this is going to be solved. Can you explain how you've done that?

@Thecave3 Kindly navigate to the esp idf folder, then navigate to \components\bt\host\bluedroid\stack\include\stack\gatt_api.h find the variable GATT_DEF_BLE_MTU_SIZE set it to whatever value you want. 

Remember it might broke connection to many other ble application,  so please note it down the path and the default value.

closed 2020-06-08 by roshanrajaratnam