nordicsemi/Android-nRF-Mesh-Library #315: in provisioning or configuration process occur gatt error

BaiduSu, 2020-05-15:
I`m provisioning a mesh device using this library, I encountered some occasional problems.  when smartphone connect the mesh provisioning service, start provisioning, most scenes can be provisioned normally,  but occasionally encountered gatt error. configuration process will occur similar situation.  test conditions: 1 android system version is 6.0; 2 peripheral device firmware is other vendor;

roshanrajaratnam, 2020-05-15:
Hi GattError can occur from time to time, I have added some delays to avoid this. This error GattError 133 is mostly happenning on Older android devices(since Android 8.0 we almost never see this happenning). There is no clear reason for this error, however it can happen when the ble resources get overwhelmed. Does this happen with Nordic mesh example firmware? I will give it a test.

BaiduSu, 2020-05-18:
I will try android 8.0

roshanrajaratnam, 2020-06-04:
@roshanrajaratnam did you manage to get around this?

roshanrajaratnam, 2020-06-09:
Closing due to inactivity!

closed 2020-06-09 by roshanrajaratnam