nordicsemi/Android-nRF-Mesh-Library #316: Will maintain the connection of multiple devices at the bottom sdk

BaiduSu, 2020-05-15:
Will maintain the connection of multiple devices at the bottom sdk? In no.nordicsemi.android.ble.BleManager.java maintain only one device connection. Will you consider maintaining multiple devices at the bottom sdk? Thanks

roshanrajaratnam, 2020-05-15:
@BaiduSu currently we don't maintain multiple connections in the sample 
mesh app and don't see us implementing this since there are other important features in the backlog. However please feel free to implement this and send us a pull request if you do, we do accept contributions :)

closed 2020-05-15 by roshanrajaratnam

philips77, 2020-05-15:
Although a single `BleManager` is responsible for one connection, you may have multipe instances of them. Have a look at [nRF Toolbox](https://github.com/NordicSemiconductor/Android-nRF-Toolbox) -> Proximity profile.