nordicsemi/Android-nRF-Mesh-Library #320: Nodes have all the same unicastaddress

Eklow02, 2020-06-04:
---

**Describe the bug**
HI, I have the problem that every provisioned node has the same unicastaddress. The code where the bug is looks like this:
```
private boolean isAddressValid(@NonNull final UnprovisionedMeshNode node) {
        final int unicast = mMeshNetwork.nextAvailableUnicastAddress(node.getNumberOfElements(), mMeshNetwork.getSelectedProvisioner());
        if (!MeshAddress.isValidUnicastAddress(unicast)) {
            throw new IllegalArgumentException("Invalid address");
        }
        if (!mMeshNetwork.getSelectedProvisioner().isAddressWithinAllocatedRange(mMeshNetwork.getUnicastAddress())) {
            throw new IllegalArgumentException("Address assigned to node is outside of provisioner's allocated unicast range.");
        }
        node.setUnicastAddress(mMeshNetwork.getUnicastAddress());
        return true;
    }
```
After the validation is done the address of the meshnetwork is used. But not the new unicastAddress. As workaround I assign a new address to the meshNetwork using nextAvailableUnicastAddress before I start with the provisioning. The sample app is using the same workaround.

**Expected behavior**
 My expectation was that the library sets the correct address to the nodes. 
Maybe I just misunderstood it. But then documentation would be helpful😄.

roshanrajaratnam, 2020-06-04:
Hi @Eklow02, how is this possible? have you provisioned multiple devices and they have the same address? Could you share a screenshot of the list of devices?

Eklow02, 2020-06-08:
Hi, in your sample app it's working because you also assign the the new address to the network before you start the provisioning. I missed this step and then every device get's the same address. My expectation was that the library is assigning the new address to the node. And because I haven't found any docu for this, I have lost a lot of time to find this issue. Maybe you can add some docu if this is the expected behavior...

roshanrajaratnam, 2020-06-08:
Ah yes. Well the provisioner has an address range and the provisioner is assigned an address by default and this will assign the rest of the unassigned addresses sequentially to the newly provisioned addresses. Yes I will include some documentation required

closed 2020-06-17 by roshanrajaratnam

saty9, 2020-12-04:
@roshanrajaratnam  I have been encountering the same issue and don't understand your response. The code block at the start of this issue still assigns the network's address not one from the provisioning nodes range. Unless there is another place the unprovisioned node gets its address assigned?