nordicsemi/Android-nRF-Mesh-Library #321: Elliptical Curve Error during Provisioning

ChadyG, 2020-06-04:
---

**Describe the bug**
Hello,  I'm following the BLE-Mesh-Node guide and am getting errors when using this Android app to provision nodes.  I'm getting the error in both the Play Store build and my own build using the current HEAD of this repo.  The specific error is below, and the full logcat is attached.
`E/MeshProvisioningHandler: Exception in PROVISIONING_PUBLIC_KEY : Invalid point coordinates`
[Android_nRF_Mesh_logcat.txt](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/4732695/Android_nRF_Mesh_logcat.txt)

I've loaded the onoff_server firmware onto a WRover module and attempting to provision it.  This process works using the iOS nRF Mesh app, but I've had no luck yet using the Android app.  I've built the onoff_server firmware using the following commit:
https://github.com/espressif/esp-idf/commit/7d75213674b4572f90c68162ad6fe9b16dae65ad

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the ESP-BLE-MESH Guide linked above.  Confgure and compile the onoff_server example for WROVER module.
2. Flash compiled firmware onto WROVER.  (I needed to set baudrate at 115200 to flash successfully)
3. Compile and install nRF Mesh app (or via Play Store) onto an Android Phone.
4. Attempt to add a node in the nRF Mesh app, app will error during provisioning with an alert box stating "Prohibited!" and above logcat.

**Expected behavior**
Android app should provision node and user can continue following Mesh Node guide.

**Platform details:**
 - Device: LG ThinQ G8
 - OS: Android 9.0
 - ESP-WROVER-KIT https://www.espressif.com/en/products/hardware/esp-wrover-kit/overview
 - esp-idf commit https://github.com/espressif/esp-idf/commit/7d75213674b4572f90c68162ad6fe9b16dae65ad
 - nRF Mesh commit https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/commit/7712ed64da449b60f9f4932b4b6f24057f3aa284

If there are any other details needed, I'll be happy to provide them.
Thanks!

roshanrajaratnam, 2020-06-08:
@ChadyG Thanks for reporting this issue, I was initially thinking if it's an MTU issue. However I see that this does not seem to be the issue. Could you see if you are facing the same with the iOS version as well? Unfortunately I don't have any hardware to test this myself and everything seems to be working fine on our end with the nRF5 Mesh Sdk and Zephyr.

roshanrajaratnam, 2020-06-08:
Looks like there seems to be a slight issue here. Let me give this a go.

closed 2020-06-08 by roshanrajaratnam

ChadyG, 2020-06-08:
Confirmed working here.

Thank you!