nordicsemi/Android-nRF-Mesh-Library #322: ConfigCompositionDataGet not working

Eklow02, 2020-06-08:
Hi, maybe you can help me with this problem. I'am using your library for my own app. After I have provisioned a device I want to sent the `ConfigCompositionDataGet`. But this request is not working. Either nothing is happening or I get into `onTransactionFailed`. In your sample App it's working. I'm pretty sure I'm doing the same steps as in the sample app. Maybe you can help me out? I have attached my logs.

[image]

roshanrajaratnam, 2020-06-08:
@Eklow02 please bear with me, I have invalidly referenced and closed your issue. I will take a look at this in a while.

Eklow02, 2020-06-08:
@roshanrajaratnam I will close this one. Don't need your help anymore. It was a problem with my ble connection ✌️

closed 2020-06-08 by Eklow02