nordicsemi/Android-nRF-Mesh-Library #333: NumberFormatException during DB import

Eklow02, 2020-06-30:
---

**Describe the bug**
After starting the App several times I get a `NumberFormatException` during the import of the db.

**To Reproduce**
Steps to reproduce the behavior:
Build and deploy the App several times via Android Studio. I don't know if this also happens in the Sample app.
The only solution of this is to clear all data of the app.

**Expected behavior**
Don't crash during db import.

**Platform details:**
 - Device:  Huawei P9, Pixel 3a
 - OS: Android 7.0, Android 10
 - Library Version 2.3.1

**Logs / Screenshots**
`
E/AndroidRuntime: FATAL EXCEPTION: AsyncTask #1
    Process: ********, PID: 21154
    java.lang.RuntimeException: An error occurred while executing doInBackground()
        at android.os.AsyncTask$3.done(AsyncTask.java:330)
        at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
        at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
        at java.util.concurrent.FutureTask.run(FutureTask.java:242)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:255)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
        at java.lang.Thread.run(Thread.java:776)
     Caused by: java.lang.NumberFormatException: For input string: "422802952"
        at java.lang.Integer.parseInt(Integer.java:524)
        at no.nordicsemi.android.mesh.transport.ElementDbMigrator.deserialize(ElementDbMigrator.java:34)
        at no.nordicsemi.android.mesh.transport.ElementDbMigrator.deserialize(ElementDbMigrator.java:20)
        at com.google.gson.internal.bind.TreeTypeAdapter.read(TreeTypeAdapter.java:69)
        at com.google.gson.internal.bind.TypeAdapterRuntimeTypeWrapper.read(TypeAdapterRuntimeTypeWrapper.java:41)
        at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.read(MapTypeAdapterFactory.java:187)
        at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.read(MapTypeAdapterFactory.java:145)
        at com.google.gson.Gson.fromJson(Gson.java:932)
        at com.google.gson.Gson.fromJson(Gson.java:897)
        at com.google.gson.Gson.fromJson(Gson.java:846)
        at no.nordicsemi.android.mesh.MeshTypeConverters.fromJsonToElements(MeshTypeConverters.java:54)
        at no.nordicsemi.android.mesh.data.ProvisionedMeshNodesDao_Impl.getNodes(ProvisionedMeshNodesDao_Impl.java:684)
        at no.nordicsemi.android.mesh.MeshNetworkDb$LoadNetworkAsyncTask.doInBackground(MeshNetworkDb.java:360)
        at no.nordicsemi.android.mesh.MeshNetworkDb$LoadNetworkAsyncTask.doInBackground(MeshNetworkDb.java:325)
        at android.os.AsyncTask$2.call(AsyncTask.java:316)
        at java.util.concurrent.FutureTask.run(FutureTask.java:237)
`

roshanrajaratnam, 2020-10-14:
@Eklow02 Have you gotten around this?

Eklow02, 2020-10-20:
Hi @roshanrajaratnam, sorry for the late reply. 
The bug is still there. But it doesn't happen that often anymore

roshanrajaratnam, 2020-10-20:
@Eklow02 I've had no luck trying to reproduce this issue. Have you tried using the latest library version?

Eklow02, 2020-10-20:
Yes I have updated to the latest library version. For me it's also hard to reproduce. The error appears irregularly

roshanrajaratnam, 2020-10-20:
I'll keep an eye on this.

TungKL, 2021-05-20:
Hello @roshanrajaratnam ,
are there any solutions to this problem yet?
I'm encountering the same bug "Caused by: java.lang.NumberFormatException: For input string: "422802952" " as described here.
I built an app using the Library version 2.4.3 and run it on a Xiaomi MI 8.

I can post my error logs if needed (although they are pretty much the same compared to the ones posted from the author of this issue)

TungKL, 2021-05-20:
Do you mean the MeshNetworkDb.java file?
Or what do you mean by DB file?

TungKL, 2021-05-24:
@roshanrajaratnam In there are the DB files I copied from the Device File Explorer in Android Studio. The app was in the same error state as described earlier. 

[mesh_network_database.zip](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/files/6533605/mesh_network_database.zip)

closed 2021-06-09 by roshanrajaratnam