nordicsemi/Android-nRF-Mesh-Library #335: Scheduler Messages

Eklow02, 2020-07-23:
Do you already know when you want to implement scheduler related messages?

philips77, 2020-07-27:
Hello,
I don't see those messages coming to the library this year. We plan to do new release in October-ish, but it's unlikely that those messages will be included. You may speed things up by submitting a PR! We will gladly accept it!

Eklow02, 2020-10-20:
Sorry for the late reply.

unfortunately my client don't let me share the code with you 😢

DevHachi, 2021-06-21:
Are there any news to this subject?

roshanrajaratnam, 2021-06-21:
Hi, we don't have any updates on this right now as we are busy with some other tasks. We gladly accept pull requests if you get around implementing them :)

roshanrajaratnam, 2021-12-17:
Hi @DevHachi thanks to @Eklow02 #448 fixes this.

closed 2021-12-17 by roshanrajaratnam