nordicsemi/Android-nRF-Mesh-Library #336: Import & Export settings

yjfkpyu, 2020-07-28:
---

**Describe the bug**
Export current settings to a json file,reset settings and  import the json file ,can see the nodes and can connect to these nodes,but can't send any data to node，unless power off and power on the node device。

**To Reproduce**
Steps to reproduce the behavior:
1.create a mesh network, connect provisioned node, read state OK.
2. Go to 'Settings'
3. Click on top right menu ，show the menu with “Import    Export Reset” 
3. Click Export，save current setttings to a json file.
4.Click Reset, and Click Import to import the saved json file.
5. Return to the Network page,can see proviisioned device, can connect to the device ,but cant not read state or send any data to the device.
6.power off the device，power on the device，now，app can send data to the device。

bradjwilson, 2020-07-28:
Have you tried changing the Unicast address of the provisioner once you import the file? Sounds like you have two provisioners with the same unicast address so the sequence numbers are off and the messages are being ignored in the mesh network.

yjfkpyu, 2020-07-28:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    
        Thanks for your reply!After importing the settings file, I did not modify the unicast address of the provisioner, and the network only had one provisioner. The same operation is normal on the iPhone mesh demo app, I can send data to the device without restarting the device after importing the settings file。
    
        
            
        
    
    
      
    
        
                    
            
                
                    
                            
                                
                            
                            
                                3173012
                            
                    
                        
                            
                                    3173012@qq.com
                            
                        
                
            
        
        
    
    
        签名由
        网易邮箱大师
        定制

Lelight-Iot, 2020-08-20:
Same Step:

roshanrajaratnam, 2020-10-14:
Hi @yjfkpyu, what @bradjwilson says is correct. iOS app comes with built in a local node that includes a provisioner. However you would still need to switch the provisioner or change the address so that the sequence numbers do not clash.

closed 2020-10-14 by roshanrajaratnam