nordicsemi/Android-nRF-Mesh-Library #339: Communication between two smartphones to create a chat

openmotion, 2020-08-17:
Hello 
can i use your library to communicate between two BLE smartphone or with google glass ?

---

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Platform details:**
 - Device: [e.g. Pixel XL]
 - OS: [e.g. Android 8.0]
 - Library Version (or Commit)

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster

philips77, 2020-08-17:
No, but you could try https://github.com/NordicSemiconductor/Android-BLE-Library if you really want to use Bluetooth LE. But there are many other ways to exchange information between Android phones. And Google Glass? Seriously?

closed 2020-08-17 by philips77

openmotion, 2020-08-17:
> No, but you could try https://github.com/NordicSemiconductor/Android-BLE-Library if you really want to use Bluetooth LE. But there are many other ways to exchange information between Android phones. And Google Glass? Seriously?

hello yes the last model of google glass