nordicsemi/Android-nRF-Mesh-Library #341: Crash db occurrence when deleting provisioned node

antoniovillanova, 2020-09-02:
---

**Describe the bug**
Randomly when deleting a provisioned node this exception occurs:

`09-02 12:23:24.361  4289 11754 E AndroidRuntime: java.lang.RuntimeException: An error occurred while executing doInBackground()
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at android.os.AsyncTask$4.done(AsyncTask.java:399)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at java.util.concurrent.FutureTask.run(FutureTask.java:271)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:289)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:919)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: Caused by: java.util.ConcurrentModificationException
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at java.util.ArrayList$Itr.next(ArrayList.java:860)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at java.util.Collections$UnmodifiableCollection$1.next(Collections.java:1089)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at androidx.room.EntityDeletionOrUpdateAdapter.handleMultiple(EntityDeletionOrUpdateAdapter.java:86)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at no.nordicsemi.android.meshprovisioner.data.ProvisionedMeshNodesDao_Impl.update(ProvisionedMeshNodesDao_Impl.java:315)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at no.nordicsemi.android.meshprovisioner.MeshNetworkDb$UpdateNetworkAsyncTask1.doInBackground(MeshNetworkDb.java:418)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at no.nordicsemi.android.meshprovisioner.MeshNetworkDb$UpdateNetworkAsyncTask1.doInBackground(MeshNetworkDb.java:385)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at android.os.AsyncTask$3.call(AsyncTask.java:378)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
09-02 12:23:24.361  4289 11754 E AndroidRuntime: 	... 4 more`

**To Reproduce**
Steps to reproduce the behavior:
1. Delete a provisioned node

**Expected behavior**
Node successfully removed from mesh db

**Platform details:**
 - OS: [e.g. Android 8.0]
 - Library Version (or Commit): meshprovisioner: 2.1.4
                                                    ble:2.1.1

antoniovillanova, 2020-09-16:
Did you manage to replicate the issue?
In which version this issue will be fixed?

Thanks.

philips77, 2020-09-16:
Hi,
The maintainer of this repo, @roshanrajaratnam, is currently on leave. This issue should be fixed in the next release, before end of this year, I guess. But I will leave it to him to investigate. Sorry, that we cannot address it faster.

roshanrajaratnam, 2020-10-14:
@antoniovillanova cannot seem to reproduce this on Android 11. Will give it a shot on Android 8.0

roshanrajaratnam, 2020-10-14:
@antoniovillanova I provisioned 3 nodes and deleted them one after the other and seemed to be working fine on Android 8.0.

closed 2020-10-14 by roshanrajaratnam

Eklow02, 2021-02-15:
I get the same error on different devices and not only when I delete a node, but also when I try to add one.
My current device is a Pixel 3a with Android 11

roshanrajaratnam, 2021-02-15:
Hey, what version of the library are you on?

Eklow02, 2021-02-15:
I'm using Version 2.4.1

roshanrajaratnam, 2021-02-15:
Could you try to upgrade and see if that helps?

Eklow02, 2021-02-15:
Yes, I will look into it and get back to you.

Eklow02, 2021-02-17:
Hi, I have updated to the latest Version `3.0.3`.  
Unfortunately  I get a similar error as described above:

`-------- beginning of crash
2021-02-17 15:07:31.238 19738-19809/de.jung.lbconnect E/AndroidRuntime: FATAL EXCEPTION: pool-4-thread-1
    Process: de.jung.lbconnect, PID: 19738
    java.util.ConcurrentModificationException
        at java.util.ArrayList$Itr.next(ArrayList.java:860)
        at java.util.Collections$UnmodifiableCollection$1.next(Collections.java:1089)
        at androidx.room.EntityDeletionOrUpdateAdapter.handleMultiple(EntityDeletionOrUpdateAdapter.java:86)
        at no.nordicsemi.android.mesh.data.ProvisionedMeshNodesDao_Impl.update(ProvisionedMeshNodesDao_Impl.java:317)
        at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$update$6(MeshNetworkDb.java:225)
        at no.nordicsemi.android.mesh.-$$Lambda$MeshNetworkDb$P_OWMUQ0TcsIRsw0eLJvpw-4dpE.run(Unknown Source:16)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:923)`

roshanrajaratnam, 2021-02-18:
Right ok now I might have a clue what's happening. This is probably because the list is being modified. Could you also give me the steps to reproduce please or is it just very random?

Eklow02, 2021-02-18:
I just want to delete a Node from the network. The error is happen very randomly. Therefore, I myself do not know exactly when this is happening. Maybe I get a notification from somewhere and the library is trying to update the network and in the same time I'm trying to delete the node

Eklow02, 2021-08-03:
Is there an update for this bug? I still get the error (version 3.1.6)