nordicsemi/Android-nRF-Mesh-Library #342: no such algorithm: ECDH for provider SC

Lelight-Iot, 2020-09-25:
---

**Describe the bug**
Method: ProvisioningPublicKeyState＃generateKeyPairs()
**log**
````
31675-31675 W/System.err: java.security.NoSuchAlgorithmException: no such algorithm: ECDH for provider SC
31675-31675 W/System.err:     at sun.security.jca.GetInstance.getService(GetInstance.java:87)
31675-31675 W/System.err:     at sun.security.jca.GetInstance.getInstance(GetInstance.java:206)
31675-31675 W/System.err:     at java.security.KeyPairGenerator.getInstance(KeyPairGenerator.java:304)
31675-31675 W/System.err:     at no.nordicsemi.android.mesh.provisionerstates.ProvisioningPublicKeyState.generateKeyPairs(ProvisioningPublicKeyState.java:105)
31675-31675 W/System.err:     at no.nordicsemi.android.mesh.provisionerstates.ProvisioningPublicKeyState.executeSend(ProvisioningPublicKeyState.java:88)
31675-31675 W/System.err:     at no.nordicsemi.android.mesh.MeshProvisioningHandler.sendProvisionerPublicKey(MeshProvisioningHandler.java:455)
31675-31675 W/System.err:     at no.nordicsemi.android.mesh.MeshProvisioningHandler.handleProvisioningWriteCallbacks(MeshProvisioningHandler.java:175)
31675-31675 W/System.err:     at no.nordicsemi.android.mesh.MeshManagerApi.handleWriteCallbacks(MeshManagerApi.java:511)
31675-31675 W/System.err:     at no.nordicsemi.android.mesh.MeshManagerApi.handleWriteCallbacks(MeshManagerApi.java:490)
````
**build.gradle**
````
buildTypes {
        release {
            minifyEnabled true
            ...
            proguardFiles getDefaultProguardFile('proguard-android-optimize.txt'), 'proguard-rules.pro'
        }
    }
````
**proguard-rules.pro**
````
...
# 
-keep class com.madgag.spongycastle.* {*;}
-keep class sun.security.jca.* {*;}
-keep class java.security.* {*;}
-keep class org.bouncycastle.* {*;}
# nineoldandroids
-dontwarn com.nineoldandroids.**
-keep class com.nineoldandroids.** { *; }
````

**To Reproduce**
Steps to reproduce the behavior:
startProvisioning
1. App run on debug is ok
2. App run on release mode is crash (minifyEnabled true)

I noticed that [issues#186](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/168). I have made sure that MeshManagerApi#initBouncyCastle() is called when the application starts.
Is there any error in my ‘proguard-rules.pro’ ?

roshanrajaratnam, 2020-10-14:
Did you have any luck with this?

Lelight-Iot, 2020-10-15:
> Did you have any luck with this?

Sorry to disturb you, I know how to solve it. 

Need
`-keep class org.spongycastle.** { *; }`

closed 2020-10-15 by Lelight-Iot