nordicsemi/Android-nRF-Mesh-Library #344: Add GenericDeltaSet for android

Kleak, 2020-10-12:
on iOS there is a `GenericDeltaSet` but not on android.
It would be great if the 2 platforms have the same capabilities.

roshanrajaratnam, 2020-10-14:
Hi @Kleak, yes as of now this is not supported on Android due to other priorities. I will try to add it in the next release.

R0m4in-dooz, 2020-11-27:
Hello @roshanrajaratnam ! We are using the library and need this feature for Android. Can we get a deadline on the future release containing GenericDeltaSet for Android ?

roshanrajaratnam, 2020-11-27:
Hey @R0m4in-dooz unfortunately I cannot promise a deadline as I have some other priorities related to mesh and I will add this on one of the upcoming releases. However feel free to send us a pull request if you decide it to implement it yourself.

R0m4in-dooz, 2020-11-27:
Allright thank you for quick response !
Well we have already some custom classes to wrap GenericMessage, but not sure if it could help you to indeed implement the feature. 
We will try to implement a workaround until a release containing this DeltaSet feature for Android is published 👍 Keep up the great work :)

ZacharyTech, 2022-08-26:
It's been two years

roshanrajaratnam, 2024-05-10:
Finally! #574 fixes this.

closed 2024-05-10 by roshanrajaratnam