nordicsemi/Android-nRF-Mesh-Library #353: Having the ability to have our own MeshNetworkCallbacks

Kleak, 2020-11-02:
As of today we only have onMeshNetworkUpdate but we lost a lot of information on why it's updated.

It would be great if we can have our on MeshNetworkCallbacks. By having this we will not lost why the MeshNetwork is updated and allow more granularity when executing functionality.

roshanrajaratnam, 2020-11-03:
Hi @Kleak, could you elaborate a little bit more on this? 

As of now every time the network is changed due to a mesh message that is sent/received you receive a callback `onNetworkUpdated()` and this is because the network is saved to the database at each operation. However there is also `MeshStatusCallbacks` that are invoked for each message received/sent meanining you have both these callbacks giving you some information everytime a network may have changed. Do you need more functionality on this?

Kleak, 2020-11-03:
in fact onNetworkUpdated is loosing lot of information.
I would like to have the event after updating the provisioner for example but i can't because i'm not sure the onNetworkUpdated is the response of my update provisioner or something else. 
But lot of other things can't be waited because we don't have the information :/

roshanrajaratnam, 2020-11-03:
Right I see, this is something I can look into but it will have to wait a little bit unfortunately. What I can do is add a callback for each database update i.e. provisioner,  app keys and etc. hope this would work for you?

Kleak, 2020-11-03:
Yes that would be ok :+1: 
Or maybe expose an other MeshNetworkCallbacks and trigger event at the same place the current one is.

roshanrajaratnam, 2020-11-03:
ok, as of now I have some other priorities but i will look in to as soon as I get some time.

closed 2026-10-05 by roshanrajaratnam