nordicsemi/Android-nRF-Mesh-Library #354: mesh packet reassembly termination implementation is wrong

haicongfu, 2020-11-13:
---

**Describe the bug**
The current implementation of reassembled mesh packet termination is based on last packet size which is wrong. It should be based on GATT_SAR_END bits in the incoming or outgoing packet. 

**To Reproduce**
Steps to reproduce the behavior:
User third party tool to generate two mesh segments, each segment equal to the MTU. 
Because the last segment length equals mtuSize, the function fails to produce the entire packet. 

In File  MeshManagerApi.java
private byte[] appendPdu(final int mtuSize, final byte[] pdu) {
 if (length < mtuSize) {
...
                final byte[] packet = mIncomingBuffer;
                mIncomingBuffer = null;
                return packet;
            }
...
}

suggested fix is 

private byte[] appendPdu(final int mtuSize, final byte[] pdu) {
 if (length < mtuSize) {
...
            final int pduType = MeshParserUtils.unsignedByteToInt(pdu[0]);
            if (pduType == ((GATT_SAR_END << SAR_BIT_OFFSET) | MeshManagerApi.PDU_TYPE_PROXY_CONFIGURATION)) {
                final byte[] packet = mIncomingBuffer;
                mIncomingBuffer = null;
                return packet;
            }
...
}

roshanrajaratnam, 2020-11-16:
@haicongfu thanks for reporting this i'll take a look at this.

roshanrajaratnam, 2021-05-25:
Sorry for the delayed reply, but is fixed and tested.

commit eb5185ea (2021-05-25): Fixes #354 an invalid gatt level segmentation issue. The lib was considering the mtu size instead of the SAR bits in the proxy pdu.
commit b85e3ed8 (2021-06-04): Fixes #354 an invalid gatt level segmentation issue. The lib was considering the mtu size instead of the SAR bits in the proxy pdu.
PR #437 (2021-06-23): Release v3.1.6
This release brings the following changes
* Specify the address of the element instead of the node address when sending message from the SensorServerActivity.java 
* Fixes #423, #418, #333 related to database migration
* Fixes #354 an invalid GATT level segmentation issue. The lib was considering the MTU size instead of the SAR bits in the proxy PDU
* Fixes #427, #406 where segmented messages to be retransmitted being get cleared. This should improve the retransmissions
* Fixes #433 where the app was crashing on Android 6.0 devices due incorrect time stamp formatting.
* Migrates [... cut]
closed 2021-05-25 by roshanrajaratnam