nordicsemi/Android-nRF-Mesh-Library #357: How do I know if the device is online？

zp0605, 2020-11-25:
How do I know if the device is online？
For example, there are 10 provisionedmeshnodes,how do I listen for every device going online or offline in my APP?

bradjwilson, 2020-11-25:
You could ping each one of the Unicast addresses for the devices. Sending a message like "proxy-get" and if you get a response then its online. Or if the devices have proxy turned on and are in the range of your app you could see them in the proxy scanner.

roshanrajaratnam, 2020-11-26:
@bradjwilson you are correct however I would suggest @zp0605 to use Heartbeat messages for this purpose. Heartbeat messages have three functions and one of them is to know if a node is still active within a mesh network. You can set them from the configuration server model to publish periodically and is recommended that the configured destination be a group address. You can find more information about this from the Heartbeat section in the Mesh profile specification.

closed 2021-02-16 by roshanrajaratnam