nordicsemi/Android-nRF-Mesh-Library #359: Deleting App Key Failed to remove from DB

harishkumar12k, 2020-12-14:
**Describe the bug**
Deleting the App Key for first time once it is created failed.
1. Add App Key(Application Key 2)
2. Delete App Key(Application Key 2)
3. Kill the App
4. Open App
5. Now in App List screen "Application Key 2" is still present, which was deleted previously.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to App Key List screen.
2. Click on button "ADD APP KEY"(Even if i click on close OR save App Key is stored in DB).
3. Now, Deleted the key. Which is success now and even UI is updated, but close the App and open again the key is not deleted.
4. I have done same process using DBinspector with Live updates checked, App Key was not deleting for first time from App(Delete was successfully once App is closed and opened again)

**Expected behavior**
1. App Key should not be saved if we tap on cross icon in "Add App Key Screen".
2. App Key should be deleted even if we did not close the App.

**Platform details:**
 - Device: Pixel 4, API 28

**Screenshots**
In Mobile it shows 1 App Key, but in DB it still shows 3 App Keys
[image]
[image]

roshanrajaratnam, 2020-12-14:
@harishkumar12k nice catch, I'll fix this in the upcoming release!

closed 2020-12-15 by roshanrajaratnam

geestore117, 2024-06-20:
I believe this issue has been reintroduced in a newer release. It can be reproduced using the original comment above. 

Library version: no.nordicsemi.android:mesh:3.3.4
Google Play Store app version: 3.3.3
  
Platform details:

Device: Pixel 7A, API 34

PR #362 (2020-12-15): App key deletion failure
* Fixes #359
* Improvements to both app key and network key management