nordicsemi/Android-nRF-Mesh-Library #369: Sensor models not provisioning

ximen, 2021-01-04:
---
Latest versions (2.4.4 and 2.4.3 checked) not properly provisioning nodes with sensor server and/or client models.
Device gets all provisioning info and successfully finishes process, but application do not add it to network list, so it is not usable.
App version 2.1.1 does such provisioning flawlessly.

**To Reproduce**
Steps to reproduce the behavior:
1. Get nRF Mesh android app v2.4.4 or 2.4.3
2. Open sample application from nRF Connect SDK 1.4.99 (sensor_client or sensor_server)
3. Flash to PCA10040 DK
4. Try to provision and add device to network

**Expected behavior**
New device added to provisioned devices list

**Platform details:**
 - Device: Samsung Galaxy S7
 - OS: Android 8.0
 - 2.4.4

roshanrajaratnam, 2021-01-05:
@ximen  Thanks for reporting this issue, I'll take a look at this and get back to you.

roshanrajaratnam, 2021-01-05:
Hi I tried to reproduce this issue and had no luck. It seems to work perfectly without any issue on version 2.4.4 on Android 8.0 on Galaxy S7 as well as on Android 10 on Pixel devices.

Do you know exactly when it might be failing as I am failing to reproduce this issue?

ximen, 2021-01-08:
Hi! Thank you for your answer. I've made some more debugging and pretty sure the problem is in option CLOCK_CONTROL_NRF_K32SRC_SYNTH in device's Kconfig - when I change 32Khz clock source from CLOCK_CONTROL_NRF_K32SRC_XTAL to CLOCK_CONTROL_NRF_K32SRC_SYNTH I see provisioning problems (dsm link loss disconnect in android logcat after sending provisioning data).

So, I think problem is on the other side - somewhere in nRF Connect SDK.

roshanrajaratnam, 2021-01-08:
Thanks for the feedback. Could you kindly create a ticket for this on DevZone please? Our tech support should be able to help you regarding the issue on the SDK.

ximen, 2021-01-08:
Surely I'll create DevZone ticket, thank you.

closed 2021-01-08 by ximen