nordicsemi/Android-nRF-Mesh-Library #370: How do I subscribe the status message from devices?

KcLeung514, 2021-01-10:
How do I subscribe the status message from devices?

Here is the scenario.
In this mesh network, there are 2 provisioners created, and 2 devices provisioned.
- Provisioner A ( unicastAddress: 1 )
- Provisioner B ( unicastAddress: 1024 )
- Device A ( unicastAddress:2 )
- Device B ( unicastAddress:5 )

We have 2 clients; mobile client A and B. They selected different provisioners, joined this network and connected one of the provisioned nodes. ( Client A selected provisioner A and Client B selected provisioner B ).

Now, **Client B** send `GenericOnOffSet( state: true )` to **Device A**, then it receive `GenericOnOffStatus( presentState: true )` from **Device A**.
What should I do to let **Client A** receive that status message from **Device A**?

I had tried solve this with following ways...

- Publication Settings
I update the publication settings of Device A and B, and set the publish address to `0xFFFF`.
It works, but Client A will receive duplicate status messages. ( one from Device A, and the another one from Device B )
I'm a little bit worry, more duplicate messages if we have more provisioned devices...

- Proxy Filter, BlackList, empty address list
I enable the BlackList Proxy Filter by send ProxyConfigSetFilterType message.
Then something weird happened.
**Client A** send `GenericOnOffSet( state: true )` to **Device B**

On **Client A**, it receives back what it sent, and cannot receive `GenericOnOffStatus( presentState: true ) from Device B`.
```
V/GenericOnOffSet(24272): State: ON
V/MeshTransport(24272): Src address: 0001
V/MeshTransport(24272): Dst address: 0005
V/MeshTransport(24272): Key: 56FA5E32A46A1EAFFF7CC6B598E1E1D5
V/MeshTransport(24272): akf: 1
V/MeshTransport(24272): aid: 6
V/MeshTransport(24272): aszmic: 0
V/MeshTransport(24272): Sequence number: 78
V/MeshTransport(24272): Access message opcode: 8202
V/MeshTransport(24272): Access message parameters: 01AA
V/AccessLayer(24272): Created Access PDU 820201AA
V/UpperTransportLayer(24272): Application nonce: 010000004E0001000500000000
V/UpperTransportLayer(24272): Encrypted upper transport pdu: 44F02636DEC8F62C
V/LowerTransportLayer(24272): Unsegmented Lower transport access PDU 4644F02636DEC8F62C
V/NetworkLayer(24272): Encryption key: E8BF842B2F258243087D4252C54E2CB2
V/NetworkLayer(24272): Privacy key: D45A6A84FDF6139D1D8A64052079B644
V/NetworkLayer(24272): Sequence Number: 00004E
V/NetworkLayer(24272): Encrypted Network payload: 1EBF51F4100C09F93A9344DD21F2E1
V/MeshManagerApi(24272): MeshNetwork pdu sent: 0x0055129632ADFF1F1EBF51F4100C09F93A9344DD21F2E1
V/MeshManagerApi(24272): Received network pdu: 0x00552DA8D3D2C34F21377D601D32BA2924495A5428E135
V/BaseMeshMessageHandler(24272): TTL for received message: 4
V/BaseMeshMessageHandler(24272): Sequence number of received access message: 78
V/NetworkLayer(24272): TTL for received message: 4
V/NetworkLayer(24272): Dst: 0x0005
D/LowerTransportLayer(24272): IV Index of received message: 0
D/LowerTransportLayer(24272): SeqAuth: 78
V/AccessLayer(24272): Received Access PDU 820201AA
V/DefaultNoOperationMessageState(24272): Unknown Access PDU Received: 820201AA
```

On **Client B**, it received `GenericOnOffStatus( presentState: true )` from Device B.

Please correct me if my concept is wrong. Thank you.

philips77, 2021-01-20:
Hello,

> What should I do to let Client A receive that status message from Device A?

What I would do:
1. Set publication on Device A to some group address. Not to 0xFFFF, as this goes to all devices, but to some custom one.
2. On Provisioner A and B add this group address to Whitelist filter.
3. Send `GenericOnOffSetUnacknowleded(state=true)` instead.

You should not receive 2 status messages on any node, as they only originate from Device A. Device B should only relay that message, changing only the TTL. As the SEQ is the same, Device A should not relay that again.

KcLeung514, 2021-01-29:
I tried the suggestion, but it seen something went wrong.

Here is the nodes created in mesh network.
- Provisioner A ( unicastAddress: 1 )
- Device A ( unicastAddress:2 )
- Device B ( unicastAddress:5 )
- Group A ( groupAddress: 0xC000 )

I updated the publication settings of Device A and B, and set the publish address to 0xC000.
When connected to provisioned node, I enabled the whiteList Proxy Filter by send ProxyConfigSetFilterType message, and added 0xC000 into address list by send ProxyConfigAddAddressToFilter message.

When `Provisioner A` send `GenericOnOffSetUnacknowleded(state=true)` to Device A, the Device A's bulb is turned on, but `Provisioner A` cannot receive `GenericOnOffStatus( presentState: true )` from `Device A`.

@philips77

KcLeung514, 2021-02-09:
The issue is solved.

Here is the story.
When I try to apply the same suggestion at iOS platform.
I saw following message from logs.
```
[Proxy] Limited Proxy Filter detected.
[Proxy] Sending SetFilterType(filterType: Whitelist) from: 0001 to: 0000
[bleMesh] Limited proxy filter detected - maxSize: (
    1
)
```
So, I think that limit come from my device's firmware.
Finally, I empty the address list before add address `0xC000`, then my app client can receive status messages from `0xC000`. Issue solved.

closed 2021-02-09 by KcLeung514

hipal31, 2022-01-29:
> Hello,
>
>
> What I would do:
>
> 1. Set publication on Device A to some group address. Not to 0xFFFF, as this goes to all devices, but to some custom one.
> 2. On Provisioner A and B add this group address [...]

Hello,
I tried same scenario but when I connect both client A and B to device A then I can receive status of device A only and when I control device B then both client not able to receive status of device B.

Here is the thing I want to test
-> I want to connect client with any device A or B but when I send GenericOnOffSetUnacknowleded to device A then  both client nodes should receive status of device A and same with device B.

can anyone help with this?

Thanks in advance