nordicsemi/Android-nRF-Mesh-Library #371: Import network problems

xhtlxa, 2021-01-12:
---

**Describe the bug**
The node cannot be set normally after importing other networks.

**To Reproduce**
Steps to reproduce the behavior:
1. Settings-Export
2. Settings-Import
3. connect

**Expected behavior**
The node can be set normally.

**Platform details:**
 - Device: Android
 - OS: 8.0
 - Library Version (2.4.3)

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster

xhtlxa, 2021-01-27:
> 设置节点是什么意思？您是否正在更改网络并期望能够访问旧网络中的节点？
I hope it can be used normally after importing the mesh. Later I found that it can be achieved by modifying the unicast address different from the previous one, but there are only a few thousand unicast addresses available and cannot be repeated with the previous ones. Is there any way I can clear the previous ones Unicast address cache

roshanrajaratnam, 2021-01-27:
I believe the problem you are having is sending messages after exporting and importing a network on a different phone?
If so the reason for this not to work is the sequence number being exported or imported for security reasons. However you can have multiple provisioners in the network for different devices.. This way you don't have to modify the unicast address of the provisioners to send messages.

xhtlxa, 2021-01-29:
> I believe the problem you are having is sending messages after exporting and importing a network on a different phone?
> If so the reason for this not to work is the sequence number being exported or [...]

I can’t use it after exporting and importing from the same mobile phone. Is there a way to directly clean up the unicast addresses that I have previously set?

xhtlxa, 2021-01-29:
> I believe the problem you are having is sending messages after exporting and importing a network on a different phone?
> If so the reason for this not to work is the sequence number being exported or [...]

Or is there a way to clear the serial number cache?

roshanrajaratnam, 2021-01-29:
If you had export >> reset >> imported again the sequence numbers are lost and the messages won't work sent from the same address because the nodes will have the last used sequence number. However, if you had just exported and imported it should just work out of the box, I will take at look at this on my end as well. 

If you want to change address of a provisioner you can do that in settings >> provisioner.

xhtlxa, 2021-02-02:
> If you had export >> reset >> imported again the sequence numbers are lost and the messages won't work sent from the same address because the nodes will have the last used sequence number. However, [...]

I can’t continue to use it after exporting and importing. It seems that the serial number has become zero, and the serial number of the node is still the previous one, so the message is filtered.

xhtlxa, 2021-02-02:
> I can’t continue to use it after exporting and importing. It seems that the serial number has become zero, and the serial number of the node is still the previous one, so the message is filtered.

Is there a way to get and modify the serial number?

roshanrajaratnam, 2021-02-02:
Here's how to set the sequence number manually https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/05ccaf31796939e8e47d138e2a91bf37c24d54bf/mesh/src/main/java/no/nordicsemi/android/mesh/transport/ProvisionedMeshNode.java#L242-L253

xhtlxa, 2021-02-02:
> Sets the sequence number 
>       * <p> 
>       *     This is only meant to be used internally within the library. 
>       *     However this is open now for users to set the sequence number manually in provisioner node

Excuse me, can this serial number overlap infinitely? Or will it automatically return to 0?

xhtlxa, 2021-02-03:
> Here's how to set the sequence number manually
> 
> https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/05ccaf31796939e8e47d138e2a91bf37c24d54bf/mesh/src/main/java/no/nordicsemi/android/mesh/transport/ProvisionedMeshNode.java#L242-L253

Why is my getSequenceNumber always -1?

roshanrajaratnam, 2021-02-03:
as I said, when exporting the network, we don't export the sequence number so when you import again it's reset.

xhtlxa, 2021-02-04:
> as I said, when exporting the network, we don't export the sequence number so when you import again it's reset. It will reset to 0 after the IV Index procedure completes.

After the sequence is reset to 0, I cannot interact with the node. How to deal with this?

roshanrajaratnam, 2021-02-04:
You will have to manually handle this in your app, this is the reason the library allows to set the sequence number manually. Sequence number is not exported due to security reasons.

closed 2021-02-04 by roshanrajaratnam

xhtlxa, 2021-02-05:
> You will have to manually handle this in your app, this is the reason the library allows to set the sequence number manually. Sequence number is not exported due to security reasons.

How do I receive the acquisition and modification sequence? The sequence I get with getSequenceNumber()  is always -1.

roshanrajaratnam, 2021-02-05:
I will be changing the default sequence number to zero, however, like I said before sequence numbers are not exported so you will have to handle it manually. What I mean by handling manually is that keep track of the the sequence numbers. Because every time you export and import the network from a json file the sequence number is not present. Hence it can be -1 or will be 0 in the future. May be you can store  it in your app.

xhtlxa, 2021-02-05:
> I will be changing the default sequence number to zero, however, like I said before sequence numbers are not exported so you will have to handle it manually. What I mean by handling manually is that keep track of the the sequence numbers. Because every time you export and import the network from a json file the sequence number is not present. Hence it can be -1 or will be 0 in the future. May be you can store it in your app.
> 
> What do you mean by acquisition and modification sequence?

What I think is getSequenceNumber()   before export and setSequenceNumber()   after import, but the value I get from getSequenceNumber()    before export is -1, is there a problem with my writing? 

sharedViewModel.getNodes().getValue().get(0).getSequenceNumber()

roshanrajaratnam, 2021-02-05:
Sequence number is incremented for each message sent, so if your node has not sent any messages it can be -1.  But I see a slight issue in the library which can be misleading. https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/ac4a019cf8079a89f0830162ca614178efa92753/mesh/src/main/java/no/nordicsemi/android/mesh/BaseMeshNetwork.java#L1038-L1040

This should allow you to get the sequence number for a particular node.

roshanrajaratnam, 2021-02-10:
@xhtlxa #395 this should help you start sending messages after exporting and importing the same network.

PR #398 (2021-02-11): version 3.0.2
* This PR brings the  following changes.
  - Feature #395 which allows re-importing an existing network and continue sending messages from the last known sequence number stored in the DB. This should allow users to re-use already stored network information and start sending messages without having to change a provisioner address on the same device. Closes #371.
  - #396 
  - #397