nordicsemi/Android-nRF-Mesh-Library #376: Release Schedule

PM-BubblyNet, 2021-01-22:
Hey! 
We see a lot of movement over here, its great, thanks! have a couple of questions:

the iOS project next release is on Beta due to incompatibility with Android project:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/releases/tag/3.0-beta1
May I ask if its planned for the next Android release to fix this?

And second Q, can I ask if there is a panned schedule for this release? when it's the target date? how close / far is?

Thanks!

philips77, 2021-01-22:
We are planning to release 3.0 final next week or in 2 weeks. We're fixing some bugs we found during testing and adding some missing features, e.g. Key Refresh Procedure. Both Android and iOS versions will be compatible with each other and will accept the JSON generated by previous versions (although will generate only new scheme).

PM-BubblyNet, 2021-01-22:
Hey @philips77 sweet, thanks! you made my day with the last 2 sentences, was worried about projects migrations from JSONs 2.X to 3, so thats nice! Thanks!

closed 2021-04-25 by PM-BubblyNet

Countrygrl00, 2021-04-25:
- [ ] - ```
``
```