nordicsemi/Android-nRF-Mesh-Library #377: Migration Issue in feature/version-3

kiyoshi-brilliant, 2021-01-23:
*CAVEAT*: I realize feature/version-3 is not released, just wanted to check if this was already known as I'm hoping this will be fixed in the final 3.0 release.

---

**Describe the bug**
When running the Example app on a build w/ feature/version-3 after having previously had 2.4.1, the following runtime error is seen:
```
2021-01-22 17:37:48.697 20509-20587/no.nordicsemi.android.nrfmeshprovisioner E/AndroidRuntime: FATAL EXCEPTION: pool-1-thread-1
    Process: no.nordicsemi.android.nrfmeshprovisioner, PID: 20509
    java.lang.IllegalStateException: Migration didn't properly handle: mesh_network(no.nordicsemi.android.mesh.MeshNetwork).
     Expected:
    TableInfo{name='mesh_network', columns={mesh_name=Column{name='mesh_name', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, last_selected=Column{name='last_selected', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, networkExclusions=Column{name='networkExclusions', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, iv_index=Column{name='iv_index', type='TEXT', affinity='2', notNull=true, primaryKeyPosition=0, defaultValue='null'}, mesh_uuid=Column{name='mesh_uuid', type='TEXT', affinity='2', notNull=true, primaryKeyPosition=1, defaultValue='null'}, partial=Column{name='partial', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, timestamp=Column{name='timestamp', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, sequence_numbers=Column{name='sequence_numbers', type='TEXT', affinity='2', notNull=true, primaryKeyPosition=0, defaultValue='null'}}, foreignKeys=[], indices=[]}
     Found:
    TableInfo{name='mesh_network', columns={mesh_name=Column{name='mesh_name', type='TEXT', affinity='2', notNull=false, primaryKeyPosition=0, defaultValue='null'}, last_selected=Column{name='last_selected', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, networkExclusions=Column{name='networkExclusions', type='TEXT', affinity='2', notNull=true, primaryKeyPosition=0, defaultValue=''{}''}, iv_index=Column{name='iv_index', type='TEXT', affinity='2', notNull=true, primaryKeyPosition=0, defaultValue='null'}, mesh_uuid=Column{name='mesh_uuid', type='TEXT', affinity='2', notNull=true, primaryKeyPosition=1, defaultValue='null'}, partial=Column{name='partial', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='1'}, timestamp=Column{name='timestamp', type='INTEGER', affinity='3', notNull=true, primaryKeyPosition=0, defaultValue='null'}, sequence_numbers=Column{name='sequence_numbers', type='TEXT', affinity='2', notNull=true, primaryKeyPosition=0, defaultValue='null'}}, foreignKeys=[], indices=[]}
        at androidx.room.RoomOpenHelper.onUpgrade(RoomOpenHelper.java:103)
        at androidx.sqlite.db.framework.FrameworkSQLiteOpenHelper$OpenHelper.onUpgrade(FrameworkSQLiteOpenHelper.java:124)
        at android.database.sqlite.SQLiteOpenHelper.getDatabaseLocked(SQLiteOpenHelper.java:417)
        at android.database.sqlite.SQLiteOpenHelper.getWritableDatabase(SQLiteOpenHelper.java:317)
        at androidx.sqlite.db.framework.FrameworkSQLiteOpenHelper$OpenHelper.getWritableSupportDatabase(FrameworkSQLiteOpenHelper.java:92)
        at androidx.sqlite.db.framework.FrameworkSQLiteOpenHelper.getWritableDatabase(FrameworkSQLiteOpenHelper.java:53)
        at androidx.room.RoomDatabase.inTransaction(RoomDatabase.java:476)
        at androidx.room.RoomDatabase.assertNotSuspendingTransaction(RoomDatabase.java:281)
        at no.nordicsemi.android.mesh.data.MeshNetworkDao_Impl.getMeshNetwork(MeshNetworkDao_Impl.java:282)
        at no.nordicsemi.android.mesh.MeshNetworkDb.lambda$loadNetwork$1(MeshNetworkDb.java:179)
        at no.nordicsemi.android.mesh.-$$Lambda$MeshNetworkDb$Lx_nOticRyYIhf1gJ5NjtiyshXQ.run(Unknown Source:16)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:919)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install and run app from tag 2.4.1 (I also added a node to ensure the database had contents)
2. Install and run app from feature/version-3
3. See Crash

**Expected behavior**
Migration is correct. 
Given how close the expected/found values are on the migration error, I imagine this is maybe an issue with the migration itself? (seems to just be complaining about the `networkExcludes` column's notNull and default values as well as the `partial` column's default value).

I didn't check if this also occurs when upgrading from later versions (2.4.2 or 2.4.3).

kiyoshi-brilliant, 2021-01-23:
I'm not that familiar with Room so I don't know if this is actually correct, but this error seems to be fixed by changing the annotations on the `BaseMeshNetwork` class:
Original:
```
     @ColumnInfo(name = "partial")
     @SerializedName("partial")
     @Expose
     boolean partial = false;
....
    @SerializedName("networkExclusions")
    @TypeConverters(MeshTypeConverters.class)
    @Expose
    protected Map<Integer, ArrayList<Integer>> networkExclusions = new HashMap<>();
```

Fixed:
```
     @ColumnInfo(name = "partial", defaultValue = "1")
     @SerializedName("partial")
     @Expose
     boolean partial = false;
....
    @NonNull
    @SerializedName("networkExclusions")
    @TypeConverters(MeshTypeConverters.class)
    @ColumnInfo(name = "networkExclusions", defaultValue = "{}")
    @Expose
    protected Map<Integer, ArrayList<Integer>> networkExclusions = new HashMap<>();
```

roshanrajaratnam, 2021-01-23:
Thanks for the report, version-3 is no released yet. However the db migration did work when I tested from 2.4.4 to 3.0. But will double check, I handle the migrations the same way for partial [here](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/feature/version-3/mesh/src/main/java/no/nordicsemi/android/mesh/MeshNetworkDb.java#L933) and networkExclusion [here](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/feature/version-3/mesh/src/main/java/no/nordicsemi/android/mesh/MeshNetworkDb.java#L943). Need to check why this hasn't worked for you though.

roshanrajaratnam, 2021-01-25:
@kiyoshi-brilliant you are correct on this issue. When migrating the Database version from 9 to 11 this issue seems to happen. I did not face this because I was migrating from 9 to 10 and 10 to 11 when I am developing. However I could fix my query or set the default value on the annotations. I would go with your solution since it's tested and working.

roshanrajaratnam, 2021-01-25:
In addition I had to add `@NonNull` annotation for the exclusions column.

PR #378 (2021-01-25): Fix database migration
* Fixes #377 - Migration was failing when upgrading database version 9 to 11.
closed 2021-01-26 by roshanrajaratnam